reynico/raspberry-noaa #1: Variables

gonzalo-, 2018-09-11:
For fuck sake be consistent with your variables, I should fire you because of this.

[https://pbot.rmdir.de/21Hyzs2X-XLFqpgNrnRuLw](url)

LamaBleu, 2018-09-11:
Yes, seems you have to create /home/pi/predict directory, and fix paths in schedule.sh.
More, nginx is set to /home/pi, totally unsecure. But the skeleton is here tks to @reynico .
I'm running from same source scripts, and customized for my own needs (I don't use http server).
At this moment I'm working on replacing rtl_fm tool[ by Luaradio script](http://wx.philandmel.com/sdr-receiver/)

reynico, 2018-09-11:
estás aburrido gonzalo?

reynico, 2018-09-11:
Thank you! I'll fix the nginx stuff and take a look at the Luaradio script.

closed 2018-09-11 by reynico