reynico/raspberry-noaa #6: Google drive.

Tom-Neverwinter, 2019-01-25:
I see you left your google drive code in there. It would be a shame if someone uploaded NSFW images.... and it auto uploaded to twitter....

reynico, 2019-01-25:
Is this any kind of threat? What's going on with you? This is my project, I work on it whenever I have some free time. If you don't like it or if you feel it is incomplete, go ahead, fork it and make it better.

Tom-Neverwinter, 2019-01-25:
I forked it. I want to build an noaa twitter system. Turns out it was probably faster to build it myself, I had to literally look everything up and make sure it was correct. then build it at least 10 times. I'm fairly certain I'm down to fstab, the lack of nofail or noauto means I'm now going to have to rebuild it again....