reynico/raspberry-noaa #7: Enchancements-Passes

kingdom1981, 2020-02-22:
South-east europe the sun.py gives always the same enchancements. Some passes has too less capture time.

reynico, 2020-02-25:
Hi! Thank you! I'll take care of it this week. Where are you located?

kingdom1981, 2020-02-25:
Greece Rhodos

kingdom1981, 2020-02-25:
Greece

reynico, 2020-02-25:
As per I tested, seems like `sun.py` is working as expected. I used 
```
obs.lat='37.97945'
obs.long='23.71622'
```
For Athens. 

https://www.timeanddate.com/astronomy/greece/athens
16:31 local time shows a 19 deg sun elevation, and the sun.py calculation shows the same for epoch time 1582641065 (14:31 GMT)

```
% python3 sun.py 1582641065
18
```

Perhaps adjusting `SAT_MIN_ELEV` helps?

kingdom1981, 2020-02-25:
% python3 sun.py 1582641065
2020-02-25 18:31:05
-31
is this normal ?
so i have to put SUN_MIN_ELEV=-30?

reynico, 2020-02-25:
Nevermind, I found the issue. Time ago I hardcoded `+3*60*60` at the date line just for timezone reasons.

Just replace
```
date = time.strftime('%Y-%m-%d %H:%M:%S', time.localtime(int(sys.argv[1])-2*60*60))
```

with
```
date = time.strftime('%Y-%m-%d %H:%M:%S', time.localtime(int(sys.argv[1])))
```

So date will be your current date for your correctly set timezone:

```
n@linux:~$ date
Tue Feb 25 19:18:30 EET 2020 <--- I set my timezone to Athens/Europe
n@linux:~$ python3 sun.py 1582641065
25
```

kingdom1981, 2020-03-04:
It's working but i changed
if [ "${SUN_ELEV}" -gt "${SUN_MIN_ELEV}" ]; then 
to 
if [ "${SUN_ELEV}" -gt 10 ]; then
in receive.sh
Also some captures are too short in time even with SAT_MIN_ELEV=30

closed 2020-04-01 by reynico

dg9bja, 2020-05-26:
Good job at all but I also had some problems with sun.py
At the moment you hard coded localtime +3*60*60 as a result you get UTC time at your place. 
I needed -2*60*60 for UTC because our difference between local time and UTC is 2 hours (Germany).
ephem uses UTC for calculations. 

I tested that with timestamp from actual at jobs to see the result.

reynico, 2020-05-26:
Thanks @dg9bja! Fixed on 81a7fd7

dg9bja, 2020-05-27:
A lot of countries here in europe changes local time end of march and in october for one hour. summer daylight changes. I don't have a problem because I know about. tzdata could help.
Another thing I tried now is https://github.com/dbdexter-dev/meteor_demod/tree/devel
There are some important changes "improved lock detection"
I got some better results with latest decode.

dk6, 2020-06-03:
Hello Nico! Project work great 4 days now! Some passes (50%) from NOOA are South to North and others are North to South so the image produced is flipped! How can we fix this automatically?