reynico/raspberry-noaa #10: Hi after install a have error on METEOR-M2 and NOAAs all ok

v222008, 2020-05-28:
```
INFO : Starting rtl_fm record
sox FAIL formats: can't open output file `/var/www/wx/meteor/audio/METEOR-M220200528-094301.wav': No such file or directory
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain set to 48.00 dB.
Tuner error set to 1 ppm.
Tuned to 137388000 Hz.
Oversampling input by: 4x.
Oversampling output by: 1x.
Buffer size: 7.11ms
Sampling at 1152000 S/s.
Output at 288000 Hz.
Signal caught, exiting!

User cancel, exiting...
INFO : Normalization in progress
sox FAIL formats: can't open input file `/var/www/wx/meteor/audio/METEOR-M220200528-094301.wav': No such file or directory
rm: cannot remove '/var/www/wx/meteor/audio/METEOR-M220200528-094301.wav': No such file or directory
INFO : Demodulation in progress (QPSK)
[FATAL]: Could not find specified file
rm: cannot remove '/var/www/wx/meteor/METEOR-M220200528-094301.wav': No such file or directory
INFO : Decoding in progress (QPSK to BMP)
Input file "/var/www/wx/meteor/METEOR-M220200528-094301.qpsk" not found!
rm: cannot remove '/var/www/wx/meteor/METEOR-M220200528-094301.qpsk': No such file or directory
ERROR : Decoding failed, either a bad pass/low SNR or a software problem

```

v222008, 2020-05-28:
I think there is not enough stake to create a directory in  ~/raspberry-noaa/common.sh
```
if [ ! -d "${METEOR_OUTPUT}/audio/" ]; then
        mkdir -m 775 -p "${METEOR_OUTPUT}/audio/"
fi
```

reynico, 2020-05-28:
Hi! is the `/var/www/wx/meteor/` directory created? Should be created automatically

v222008, 2020-05-28:
All works fine after add

```
if [ ! -d "${METEOR_OUTPUT}/audio/" ]; then
        mkdir -m 775 -p "${METEOR_OUTPUT}/audio/"
fi
```

closed 2020-05-28 by v222008

reynico, 2020-05-30:
Thanks sir, I've commited the change  428cb2b