reynico/raspberry-noaa #12: Automatic configuration of RTLSDR-Dongle fails

df6dbf, 2020-06-11:
In fresh Installations the automatic config of RTLSDR-Dongles failed. Have to configure the Device by Hand. Rest of Installation Script work fine!

Kind Regards, Holger

reynico, 2020-06-11:
Hi Holger! hope you're doing well.
Any log output of the error?

df6dbf, 2020-06-11:
Sorry, have no log saved. I am familar with RTLSDR so i configured by hand. But a little suggestion: Take a look here: https://github.com/steve-m/kalibrate-rtl it would be nice to have a automated frequency fix of the SDR-Dongles. :) Kind Regards, Holger

reynico, 2020-06-13:
Thanks Holger! I'll give a try. Will close this issue for now, please feel free to reopen if anything else happens with the setup

closed 2020-06-13 by reynico