reynico/raspberry-noaa #13: Change of Text in Twitter Messages

df6dbf, 2020-06-11:
Is it possible to make the Text User configurable?
I am live here in Germany and my wish is to change the Text 

Imagen satelital: NOAA 15 11-06-2020 18:29. Elevacion maxima: 24 grados. #NOAA #weather #argentinaimagenes #noaasatellite #clima #wxtoimg #raspberrypi #argentina #argentinasat

to German or English.

Kind Regards,

Holger

reynico, 2020-06-11:
Sure thing!

dk6, 2020-06-12:
> Is it possible to make the Text User configurable?
> I am live here in Germany and my wish is to change the Text
> 
> Imagen satelital: NOAA [15 11-06-2020](tel:15 11-06-2020) 18:29. Elevacion maxima: 24 grados. #NOAA #weather #argentinaimagenes #noaasatellite #clima #wxtoimg #raspberrypi #argentina #argentinasat
> 
> to German or English.
> 
> Kind Regards,
> 
> Holger

Open and edit the post.py

sudo nano post.py

df6dbf, 2020-06-12:
I have found the part in the code and changed it. But i am an "Hardware-Guy" ask me about Antennas and RF-Measurement but not to so magical things like coding! :)
Have checked the directory structure as adviced, everything fine here. But my Antenne is not on her final place and i have to add the preamp to.

reynico, 2020-06-13:
I'll add some configuration option so you don't need to dive into setup files to change the text!

> as my installation didn't create the audio directory and thus couldn't save the .wav files anywhere to be decoded.

already fixed 😄