reynico/raspberry-noaa #14: Date incorrect

Eric738, 2020-06-13:
GM 
i noticed when i send in cmd mode : ./schedule_meteor.sh 
date: date « @ » incorrecte
is it a correct answer ?

i copied /paste latest "schedule_meteor.sh" but i had same answer , how i can fix ?

thank
Eric

dk6, 2020-06-13:
If you run ./schedule.sh you have noaa and meteor passes. Did you try it?

Eric738, 2020-06-13:
yea , I wanted to verify that the meteor pass were taken into account (and asked me why a separate file)

Eric

dk6, 2020-06-13:
Verify with command:>
atq
Delete a job:>
atrm <number of job>

Download an app to verify the sat passes like SatSat or pc software Orbitron etc.. remember to put your location coordinates!!

dk6, 2020-06-13:
Verify with command:>
atq
Delete a job:>
atrm <number of job>

Download an app to verify the sat passes like SAT or pc software Orbitron etc.. remember to put your location coordinates!!

Eric738, 2020-06-13:
ok for tip, i will do, thank

reynico, 2020-06-13:
Hi Eric, pretty much like either the date of your Raspberry is off (maybe an incorrect TZ) or just no available passes as for that day. Anyways that's an error that needs a fix. Could you please append

```
set -x
```
at the second line of the `schedule_meteor.sh`  script? Then paste the output here so we can take a look into it

tomamdy, 2020-06-25:
I have a receiving problems, only one line in firefox,, that's all . What could be te problem ?

reynico, 2020-07-16:
Closing by inactivity

closed 2020-07-16 by reynico