reynico/raspberry-noaa #16: Good images cropped to narrow band

ant-thomas, 2020-07-07:
Not sure if this is a feature or an issue. Or I'm more so looking for a bit of info/advice.
I've made a pretty janky indoor dipole antenna with some 1.5mm2 copper cable, also added an amplifier. Was previously using the amp for an ADS-B setup, just changed the antenna.
Not tuned in any way, antenna length is about right.

[image]

Most passes result in some signal, but generally poor.
Occasionally there's a pass which is very good, but the output image is cropped to what looks like the good section, rather than the full length of the pass/recording.

Is this supposed to happen to remove any noise across the "bad" area?
Unsurprisingly the good area has my location near enough centered.

Eg....
[image]

Great to see the setup works, but would be interested to see the full pass on "good" receive.

ant-thomas, 2020-07-07:
Interestingly just had a full pass with the good bit in the middle, along with the rest noisy....
Will eventually look at a better antenna, but would be good to know if there's a setting to be changed to prevent the crop?

[image]

reynico, 2020-07-07:
Hi Thomas! Sure, append `-A` to the wxtoimg line

```
         -A         all (do not cut top and bottom low signal sections)
```

So it'll end up like

```
        /usr/local/bin/wxtoimg -A -o -m "${NOAA_HOME}/map/${3}-map.png" -e "$i" "${NOAA_AUDIO}/audio/${3}.wav" "${NOAA_OUTPUT}/image/${FOLDER_DATE}/${3}-$i.jpg"
```

ant-thomas, 2020-07-07:
Fantastic, thank you!

closed 2020-07-07 by ant-thomas