reynico/raspberry-noaa #17: how install new release please ?

Eric738, 2020-07-10:
GE
i would like test new release but i dont understand how install..can you add one little info . Thank in advance

eric

reynico, 2020-07-10:
Hi Eric, there is no way to do an incremental install yet so you need to delete (or move to another directory) the config files generated by install.sh, as well as the installed packages.

Eric738, 2020-07-11:
thank you it is clearer for the beginners, but I meet problems of rights which I did not have before and I have difficulty.
example after ./schedule.sh:
  mkdir: unable to create directory "/ var / ramfs": Permission not granted
mkdir: unable to create directory "/ var / ramfs": Permission not granted
METEOR-M2 20200711-220345 68
warning: commands will be executed using / bin / sh
but i found soluce i I entered all the command lines manually and the error messages disappeared, so apparently during the install.sh all the lines are not taken into account except error on my part
----------
dont want install numpy 1.18.1...i install 1.9 latest
------
i received a big wave file after meteor pass; but to have taken the directories correctly because no image is displayed the meteor_demod directory must be in / home / pi / raspberry-noaa or in / home / pi / raspberry-noaa / software? which would explain why no image appeared can be

thank for your help
Eric

reynico, 2020-07-16:
Hi, for positive values just write 2.

Eric738, 2020-07-16:
thank for info

closed 2020-07-17 by reynico