reynico/raspberry-noaa #18: Allow bias-tee config on initial setup or another config file

cocus, 2020-07-12:
Hi! Another fellow Argentinean here. I'm using this code, which works pretty well, but I had to modify it in order to use the bias tee, since I have the SAWbird+ NOAA which requires it to work. I know the change is pretty much straightforward on the `receive.sh` script, but maybe that could be an option to be asked on install-time.

reynico, 2020-07-12:
Hi Santi!
Sure thing, I'll wrap that at install time so you don't need to dig into the files to enable it.

reynico, 2020-07-12:
https://github.com/reynico/raspberry-noaa/pull/19

tomamdy, 2020-07-13:
I have tried to install again, but no bias-tee possibilities.

reynico, 2020-07-13:
Hi! Have you checked out the branch `enable_biastee` before running install.sh?

tomamdy, 2020-07-13:
Which branch do You advice ?

tomamdy, 2020-07-14:
> 
> 
> Hi! Another fellow Argentinean here. I'm using this code, which works pretty well, but I had to modify it in order to use the bias tee, since I have the SAWbird+ NOAA which requires it to work. I know the change is pretty much straightforward on the `receive.sh` script, but maybe that could be an option to be asked on install-time.
> Even a simple parameter on the `~/.noaa.conf` could be used to determine this.
> An appropriate bias-tee enabler argument is: `-T` (or at least, that worked for me)

Which branch are You using ?

cocus, 2020-07-14:
I tried his new branch `enable_biastee` which is the one from the PR #19

tomamdy, 2020-07-14:
But it doesn't worked for me.

reynico, 2020-07-14:
uhm, you may need to either delete the `~/.noaa.conf` file to copy the new template or run the next command to add the required line
```
echo "BIAS_TEE=\"-T\"" >> ~/.noaa.conf
```

tomamdy, 2020-07-14:
Sorry, but nothing. I really don't understand this.

cocus, 2020-07-14:
Which hardware do you have?
Please run a `cat ~/.noaa.conf` and copy the output. Also do a `cat ~/raspberry-noaa/receive.sh` just to check if you got the changes.

tomamdy, 2020-07-14:
> Links:

I use a raspberry pi 3

NOAA_HOME=/home/pi/raspberry-noaa
NOAA_OUTPUT=/var/www/wx
METEOR_OUTPUT=/var/www/wx/meteor
NOAA_AUDIO=/var/ramfs
SAT_MIN_ELEV=30
METEOR_MIN_ELEV=30
SUN_MIN_ELEV=10
LOG_LEVEL=DEBUG
LAT=52.7166667
LON=5.2833333
BIAS_TEE="-T"
BIAS_TEE="-T"
BIAS_TEE="-T"
BIAS_TEE="-T"
BIAS_TEE="-T"
BIAS_TEE="-T"
BIAS_TEE="-T"

pi@raspberrypi:~ $ cat ~/raspberry-noaa/receive.sh
#!/bin/bash

## import common lib
. "$HOME/.noaa.conf"
. "$HOME/.tweepy.conf"
. "$NOAA_HOME/common.sh"

## pass start timestamp and sun elevation
PASS_START=$(expr "$5" + 90)
SUN_ELEV=$(python3 "$NOAA_HOME"/sun.py "$PASS_START")

if pgrep "rtl_fm" > /dev/null
then
     log "There is an existing rtl_fm instance running, I quit" "ERROR"
     exit 1
fi

# $1 = Satellite Name
# $2 = Frequency
# $3 = FileName base
# $4 = TLE File
# $5 = EPOC start time
# $6 = Time to capture
# $7 = Satellite max elevation

timeout "${6}" /usr/local/bin/rtl_fm -f "${2}"M -s 60k -g 50 -p 55 -E 
wav -E deemp -F 9 - | /usr/bin/sox -t raw -e signed -c 1 -b 16 -r 60000 
- "${NOAA_AUDIO}/audio/${3}.wav" rate 11025

if [ ! -d "{NOAA_OUTPUT}/image/${FOLDER_DATE}" ]; then
     mkdir -m 775 -p "${NOAA_OUTPUT}/image/${FOLDER_DATE}"
fi

if [ "${SUN_ELEV}" -gt "${SUN_MIN_ELEV}" ]; then
     ENHANCEMENTS="ZA MCIR MCIR-precip MSA MSA-precip HVC-precip 
HVCT-precip HVC HVCT"
else
     ENHANCEMENTS="ZA MCIR MCIR-precip"
fi

/usr/local/bin/wxmap -T "${1}" -H "${4}" -p 0 -l 0 -o "${PASS_START}" 
"${NOAA_HOME}/map/${3}-map.png"
for i in $ENHANCEMENTS; do
     /usr/local/bin/wxtoimg -o -m "${NOAA_HOME}/map/${3}-map.png" -e "$i" 
"${NOAA_AUDIO}/audio/${3}.wav" 
"${NOAA_OUTPUT}/image/${FOLDER_DATE}/${3}-$i.jpg"
     /usr/bin/convert -quality 90 -format jpg 
"${NOAA_OUTPUT}/image/${FOLDER_DATE}/${3}-$i.jpg" -undercolor black 
-fill yellow -pointsize 18 -annotate +20+20 "${1} $i ${START_DATE}" 
"${NOAA_OUTPUT}/image/${FOLDER_DATE}/${3}-$i.jpg"
done
if [ -n "$CONSUMER_KEY" ]; then
     if [ "${SUN_ELEV}" -gt "${SUN_MIN_ELEV}" ]; then
         python3 "${NOAA_HOME}/post.py" "$1 ${START_DATE}" "$7" 
"${NOAA_OUTPUT}/image/${FOLDER_DATE}/$3-MCIR-precip.jpg" 
"${NOAA_OUTPUT}/image/${FOLDER_DATE}/$3-MSA-precip.jpg" 
"${NOAA_OUTPUT}/image/${FOLDER_DATE}/$3-HVC-precip.jpg" 
"${NOAA_OUTPUT}/image/${FOLDER_DATE}/$3-HVCT-precip.jpg"
     else
         python3 "${NOAA_HOME}/post.py" "$1 ${START_DATE}" "$7" 
"${NOAA_OUTPUT}/image/${FOLDER_DATE}/$3-MCIR-precip.jpg" 
"${NOAA_OUTPUT}/image/${FOLDER_DATE}/$3-MCIR.jpg"
     fi
fi

rm "${NOAA_AUDIO}/audio/${3}

cocus, 2020-07-15:
For some reason, your receive.sh is not up to date to the branch we've mentioned. Try this
`cd ~/raspberry-noaa
git fetch && git checkout enable_biastee`

tomamdy, 2020-07-16:
I used cd ~/raspberry-noaa
git fetch && git checkout enable_biastee and he has written the cammand in noaa.conf

but is that enough for starting the bias tee  ?

Because nothing changed in receive.sh and  receive_meteor.sh

I still believe that there is an eroor in the install.sh script is downloaded again . Nothing like

reynico, 2020-07-16:
Hi @tomamdy, after git fetch & checkout did you ran the install.sh script again? That way it'll replace the BIAS_TEE var value. You can also run `sed -i -e "s/enable_bias_tee/-T/g" "$HOME/.noaa.conf"` by now to avoid re running the whole install script.

How does your receive.sh line 28  looks like?

tomamdy, 2020-07-16:
timeout "${6}" /usr/local/bin/rtl_fm ${BIAS_TEE} -f "${2}"M -s 60k -g 50 -p 55 -E wav -E deemp -F 9 - | /usr/bin/sox -t raw -e signed -c 1 -b 16 -r 60000 - "${NOAA_AUDIO}/audio/${3}.wav" rate 11025
this looks better, but I only know it tomorrow. Thank You and Santi!  in advance. I cross my fingers.

reynico, 2020-07-17:
Cool! I'll close this issue for now. Feel free to reopen it

closed 2020-07-17 by reynico

tomamdy, 2020-07-17:
ut it doesn't works. Is it possible for me to test directly the bias_tee_enable, waiting for severall hours after a new install is not so pleasant.

tomamdy, 2020-07-17:
For the .... time I have installed the script again and now it asked for enable bias_tee . That looks better. Could it be that install.sh was not changed, because nw I see that it is another install.sh.

PR #19 (2020-07-17): Setup bias-tee during install
closes #18