reynico/raspberry-noaa #20: Nginx webserver - Audiotab "Error 404"

Crosswalkersam, 2020-07-12:
Hi! I just set up my Raspberry Pi 2B+ with the Nooelec neSDR SMARTV4. I already recorded 4 NOAA passes and I am able to view the created pictures on the webserver. I noticed a lot of noise, which surprised me since I got way better images when manually recording a pass with SDR#. Thats why I wanted to listen to the corresponding .wav file. However, I am not able to access the "Audio" tab on the webserver. It just says "404 Not Found". What do I have to do to get this tab working or where can I find the .wav files when using SFTP?

Thanks in advance,
sam

reynico, 2020-07-13:
Hi Sam!
Replace 
```
NOAA_AUDIO=/var/ramfs
```
by
```
NOAA_AUDIO=/var/www/wx/audio
```
on `/home/pi/.noaa.conf`

and 
```
mkdir -p /var/www/wx/audio
```

Then remove or comment the last line of `receive.sh` 
```
rm "${NOAA_AUDIO}/audio/${3}.wav" # Comment or delete this
```

Crosswalkersam, 2020-07-14:
Thank you very much! The Audiotab is working now :)

closed 2020-07-14 by Crosswalkersam