reynico/raspberry-noaa #21: Working check

HalloweenMagic, 2020-07-15:
How can I check that everything it's ok after ./install.sh?

I would like to understand if my antenna receives and if the software starts when there are passages.
Thanks :D

reynico, 2020-07-15:
Reception tests could be done using socat. Adjust 90.3M to tune a local broadcast FM
Raspberry pi:
```
rtl_fm -f 90.3M -s 256k -g 45 -p 55 -E deemp -F 9 - | sox -traw -r256k -es -b16 -c1 -V1 - -tmp3 - |  socat -u - TCP-LISTEN:8073
```

*nix computer:
```
nc raspberry.pi.ip.here 8073 | play -t mp3 -
```

For schedule testing just run `/home/pi/raspberry-noaa/schedule.sh`, that'll schedule the passes for that day. Then you can see them with  `atq`

```
pi@wx:~ $ atq
306	Wed Jul 15 18:58:00 2020 a pi
310	Wed Jul 15 20:31:00 2020 a pi
304	Wed Jul 15 20:38:00 2020 a pi
308	Wed Jul 15 22:13:00 2020 a pi
```

And the details with `at -c <id>`
```
pi@wx:~ $ at -c 306
#!/bin/sh
# atrun uid=1000 gid=1000
# mail pi 0
umask 22
PWD=/home/pi; export PWD
LOGNAME=pi; export LOGNAME
HOME=/home/pi; export HOME
LANG=en_US.UTF-8; export LANG
SHLVL=1; export SHLVL
PATH=/usr/bin:/bin; export PATH
cd /home/pi || {
	 echo 'Execution directory inaccessible' >&2
	 exit 1
}
/home/pi/raspberry-noaa/receive.sh "NOAA 19" 137.1000 NOAA1920200715-185819 /home/pi/raspberry-noaa/predict/weather.tle 1594850299 962 68
```

HalloweenMagic, 2020-07-15:
@reynico many thanks :)

closed 2020-07-16 by reynico