reynico/raspberry-noaa #22: Directories Errors

HalloweenMagic, 2020-07-16:
Hello!
After the run of autoinstall, there are some issues with directories.
For example: audio directory doesn't exist and METEOR image are saved into NOAA and inside METEOR there aren't anything.

I hope I can help you with this report :)

[image]
[image]
[image]

tomamdy, 2020-07-16:
Yes, You are right . I like this software, but it has some issues.

reynico, 2020-07-16:
Hi! Thank you, I'll push an update to fix this.

HalloweenMagic, 2020-07-16:
> Hi! Thank you, I'll push an update to fix this.

Thanks a lot! 
There is another problem with predict: if you try to use it, you can enter immediatly inside "Configuration Mode", without possibility to choosing options from the menu

reynico, 2020-07-17:
Hi,
I've added a boolean to either store or not the received audio. git fetch && git pull to grab the latest changes and then do a 
```
echo "DELETE_AUDIO=false" >> ~/.noaa.conf
```

closed 2020-07-17 by reynico

Crosswalkersam, 2020-07-26:
@HalloweenMagic 

I had that issue with pridict aswell. to solve it, simply run:

`cd ~/.predict`

and then download the missing files from github:

`wget https://raw.githubusercontent.com/kd2bd/predict/master/default/predict.db`

`wget https://raw.githubusercontent.com/kd2bd/predict/master/default/predict.tle`

`wget https://raw.githubusercontent.com/kd2bd/predict/master/default/predict.qth`

after that, run `predict` again, it should work now.

PR #25 (2020-07-17): Toggle audio save
Closes #22