reynico/raspberry-noaa #23: No capture

MammaMia2020, 2020-07-16:
Hi and thanks for this script!
This morning I try to receive METEOR, but despite it having passed over my head, I don't know if the software has started receiving. Yesterday everything worked, not this morning. The only image produced is from NOAA, but only a small piece.

Checking with ORBITRON, I don't find myself with the dates and times.
Latitude and longitude is correct.

If I start "predict", only the page opens to configure latitude and longitude and the menu for the other functions does not appear.

Is it a bug?

[image]

This is the NOAA image cutted

[image]

And this is the problem with predict

[image]

I just see this screen:

[image]

reynico, 2020-07-17:
Hi, 
there's a check here: https://github.com/reynico/raspberry-noaa/blob/master/receive_meteor.sh#L12-L15 to avoid night passes of Meteor since the IR radiometers seems to be broken.  If you want to receive and decode all passes just comment or delete the line 14 (exit 0)

closed 2020-07-17 by reynico

Crosswalkersam, 2020-07-26:
@MammaMia2020 

To fix the issue with predict, simply do this:

`cd ~/.predict`

and then download the missing files from github:

`wget https://raw.githubusercontent.com/kd2bd/predict/master/default/predict.db`

`wget https://raw.githubusercontent.com/kd2bd/predict/master/default/predict.tle`

`wget https://raw.githubusercontent.com/kd2bd/predict/master/default/predict.qth`

after that, run `predict` again, it should work now.