reynico/raspberry-noaa #28: Bad image quality / Enable AGC

Crosswalkersam, 2020-07-26:
Hi, I am using the Raspberry Pi 4 2Gb with a Nooelec neSDR SMARTEE and a Nooelec NOAA Sawbird filter. I am struggeling to get a good image. With the Pi 2B+ and the "normal" neSDR SMART the images were great. I guess it has to do something with the gain. I tried everything from 10dBm to 50dBm in the `receive.sh` file. Do you know why that is? Maybe Automatic gain control would help, but how do I enable it?

[image]

Many thanks,
Sam

reynico, 2020-07-26:
Hi Sam, just remove the `-g` param from rtl_fm, that enables the AGC. I'm using a FM trap filter and a RTL-SDR v3 receiver, been trying for a while with different amplifiers but had no luck at all, best images were captured just with the receiver and the filter.

tomamdy, 2020-07-27:
I have the same problem, but after removing -g 50 , rtl-test gives the gain automatic.
But with -g 50 too to be automatic . So, it seems that nothing changed .

Crosswalkersam, 2020-07-29:
Thanks!
With AGC enabled and without LNA it works fine. But I have no idea why it refuses to work with the LNA. I tried different gain settings,supplied the LNA with USB or Bias-T but none of that did the trick. Even different antenna types(I tried QFH,DCA and Yagi).

closed 2020-08-02 by reynico