reynico/raspberry-noaa #29: No Meteor images

tomamdy, 2020-08-02:
I get no images in the script. What is wrong.?

METEOR-M220200725-091111.qpsk                      25-Jul-2020 10:29           128147320
METEOR-M220200725-091111.wav                       25-Jul-2020 10:21           341759360
METEOR-M220200726-085112.qpsk                      26-Jul-2020 09:15           132082368
METEOR-M220200726-085112.wav                       26-Jul-2020 09:08           352245120
METEOR-M220200727-083115.qpsk                      27-Jul-2020 08:56           134060214
METEOR-M220200727-083115.wav                       27-Jul-2020 08:48           357520768

reynico, 2020-08-02:
Hi, have you checked the `/var/www/wx/image` folder?

tomamdy, 2020-08-03:
Ofcourse
[image]
[image]

alexnxt20, 2020-08-05:
Can you post your ".noaa.conf" file located in /home/pi/?

I had a problem similar to this. The ./receive_meteor.sh script breaks if the DELETE_AUDIO variable is not either "true" or left blank. If you type DELETE_AUDIO=false in the .nooa.conf file, it will not provide images, if you want to save audio, leave the DELETE_AUDIO field blank.

tomamdy, 2020-08-05:
Here the file, I hope its gives the result.

NOAA_HOME=/home/pi/raspberry-noaa
NOAA_OUTPUT=/var/www/wx
METEOR_OUTPUT=/var/www/wx/meteor
NOAA_AUDIO=/var/ramfs
SAT_MIN_ELEV=30
METEOR_MIN_ELEV=30
SUN_MIN_ELEV=10
LOG_LEVEL=DEBUG
LAT=change_latitude
LON=change_longitude
BIAS_TEE="enable_bias_tee"
DELETE_AUDIO=true

tomamdy, 2020-08-08:
Now it works again. I only have installed the scripts again.

closed 2020-08-08 by tomamdy