reynico/raspberry-noaa #33: Cant receive any pciture from sdr.

shazlia89, 2020-08-17:
Hello everyone,sorry for my bad english. Im just wondering why i cant receive any picture from sdr. Currently i lived in Malaysia. I think most likely i configured the predict program wrongly. and i think i have a wrong position for the satellite. 
[image]
[image]

 May i know what causes this problem? Thanks alot !!

reynico, 2020-08-17:
Hi @shazlia89 !

Those images belongs to the map that WxToIMG uses to overlay received images. Have you checked the `/var/www/wx/image` directory? Take a look through the browser at http://your.raspberry.pi.ip/image/

reynico, 2020-09-13:
Closing by inactivity

closed 2020-09-13 by reynico