reynico/raspberry-noaa #34: how can I check if the SDR is up n'n runing?

DH0GME, 2020-08-20:
First: an absolute cool project - congrat's !!!
I've set up the hardware (PI4 2GB) with an Nooelec SDR . fresh rasbian, activated VNC. device seems to work in my WIFI
enwironment, Twitter API tokens set - so good so fine. But how can I check online, if and how the RX is working while
a NOAA is actually passing over me? Sorry I'm generation "windows", just making my first steps on penguin island...
73!
Elmar | DH0GME

reynico, 2020-08-20:
Hi Elmar!
The raspberry-noaa workflow is kinda simple
`crontab -l` will show you the scheduler setup:
```
1 0 * * * /home/pi/raspberry-noaa/schedule.sh
```
That means the script `schedule.sh` will run at 00:01 every day. The script downloads the kepler elements for each satellite we're interested (AKA noaa and meteor) and then builds a list of jobs that'll run during the day.

using `atq` shows the jobs that are scheduled for today, so each line is a satellite pass:
```
597	Thu Aug 20 18:39:00 2020 a pi
599	Thu Aug 20 21:51:00 2020 a pi
601	Thu Aug 20 20:24:00 2020 a pi
595	Thu Aug 20 20:22:00 2020 a pi
```

if you're interested in some particular pass, then do a `at -c 601` where 601 is the id for the job. That shows some detailed information such as the satellite name, frequency, pass duration and max sat elevation

After a satellite pass is done and decoded open a browser to http://your.raspberry.pi.ip.address/ and navigate through the images folder, there you'll see your pictures.

DH0GME, 2020-08-20:
Ni Nico,
ok TU - that's explains the first part of the story - I haven't run the device over the night!
But is it possible to "see" live, what the RX is doing while an image is in progress?
Just for checking the antenna and rest of HF frontend is working properly...
Best regards
Elmar | DH0GME

reynico, 2020-08-20:
Elmar, I wrote a small script to test the node reception. Please check: https://github.com/reynico/raspberry-noaa/blob/master/WORKING.md

SSH into your Raspberry PI, then
```
cd raspberry-noaa/
git fetch && git pull
```

to grab the script. Then you can run `./test_reception.sh 90.3` where 90.3 is the frequency you want to tune to test reception. I recommend any broadcast FM near you.

---

Regarding the Raspberry PI working hours, this is not a problem at all, but you will hace to execute `/home/pi/raspberry-noaa/schedule.sh` every time you power-up your Raspberry PI (Can be automated during the startup process)

DH0GME, 2020-08-20:
Hi Nico, tnx for your fast response - I'll try asap !!

DH0GME, 2020-08-23:
ups . sorry :-) - yep now it works - have to get my antenna a liitle higher and the coax a bit shorter....
TNX fer this project and your support!!!

reynico, 2020-08-23:
glad it worked out well!

closed 2020-08-23 by reynico