reynico/raspberry-noaa #36: Tweepy.conf

TheFarmer85, 2020-08-28:
Hi, sorry but i can't understand how doing to share pictures of my 'raspberry noaa' by Twetter....i am a beginer with tweeter..but i have now, an account of tweeter developer key and tokken key and token secret, i write it in the tweepy.conf but after what i do !???
(I am french, sorry for my english)
Many thanks.
Mickaël

reynico, 2020-08-28:
Hi Mickaël!
If you wrote the consumer key, consumer secret, access token key and access token secret to `/home/pi/.tweepy.conf` that's all you need to get it working. The next satellite pass will post to twitter.

TheFarmer85, 2020-08-30:
Hi nico, Thank you for answering.That don't work, no pictures on my account of Tweeter !Must be kept this : ='xxxx' after = or just =xxxx in tweepy.conf ?!Many thanks.Envoyé de mon Galaxy A40 Orange
-------- Message d'origine --------De : Nico <notifications@github.com> Date : 29/08/2020  00:43  (GMT+01:00) À : reynico/raspberry-noaa <raspberry-noaa@noreply.github.com> Cc : TheFarmer85 <lemarouin@orange.fr>, Author <author@noreply.github.com> Objet : Re: [reynico/raspberry-noaa] Tweepy.conf (#36) 

Hi Mickaël!

If you wrote the consumer key, consumer secret, access token key and access token secret to /home/pi/.tweepy.conf that's all you need to get it working. The next satellite pass will post to twitter.

reynico, 2020-08-30:
Hi!
Format should look like
```
export CONSUMER_KEY="3m3VR..."
export CONSUMER_SECRET="3amj..."
export ACCESS_TOKEN_KEY="103..."
export ACCESS_TOKEN_SECRET="kXpS..."
```

surfrod, 2020-09-10:
I applied for a developer account and was approved.

However, the *only* credentials I received are:

API Key, API Secret Key and Bearer Token.

How do these map into tweepy.conf??

surfrod, 2020-09-12:
I am getting all sorts of errors trying to upload images to my twitter account.

If I past upload.twitter.com into a browser, I get 

“The Twitter REST API v1 is no longer active. Please migrate to API v1.1. https://dev.twitter.com/docs/api/1.1/overview.“

The errors from the job include:

 File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 638, in 
urlopen
    _stacktrace=sys.exc_info()[2])
  File "/usr/lib/python3/dist-packages/urllib3/util/retry.py", line 398, in incr
ement
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='upload.twitter.com',
 port=443): Max retries exceeded with url: /1.1/media/upload.json (Caused by New
ConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0xb58f3d1
0>: Failed to establish a new connection: [Errno -3] Temporary failure in name r
esolution'))

......

NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0xb58f
3d10>: Failed to establish a new connection: [Errno -3] Temporary failure in nam
e resolution'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/raspberry-noaa/post.py", line 23, in <module>
    res = api.media_upload(filename)
  File "/usr/local/lib/python3.7/dist-packages/tweepy/api.py", line 242, in medi
a_upload
    )(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/tweepy/binder.py", line 252, in _
call
    return method.execute()

  File "/usr/local/lib/python3.7/dist-packages/tweepy/binder.py", line 192, in e
xecute
    six.reraise(TweepError, TweepError('Failed to send request: %s' % e), sys.ex
c_info()[2])
  File "/usr/lib/python3/dist-packages/six.py", line 692, in reraise
    raise value.with_traceback(tb)
  File "/usr/local/lib/python3.7/dist-packages/tweepy/binder.py", line 190, in e
xecute
    proxies=self.api.proxy)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 533, in reque
st
    resp = self.send(prep, **send_kwargs)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 516, in send
    raise ConnectionError(e, request=request)
tweepy.error.TweepError: Failed to send request: HTTPSConnectionPool(host='uploa
d.twitter.com', port=443): Max retries exceeded with url: /1.1/media/upload.json
 (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection obje
ct at 0xb58f3d10>: Failed to establish a new connection: [Errno -3] Temporary fa
ilure in name resolution'))
INFO : Deleting audio files
?

surfrod, 2020-09-12:
Any idea where the "spanish" is being posted? I changed the verbiage in post.py, but last tweet still had an argentinian flag and spanish.

reynico, 2020-09-12:
Closing due to inactivity

closed 2020-09-12 by reynico

pineapplee85, 2020-12-01:
Hi,

I too am having issues with posting to twitter.

I have my config and keys in the right place but still no posting?

Thanks