reynico/raspberry-noaa #37: Time zone

96611229, 2020-08-30:
Hi!

Absolutely great work on this! I've just set it up but i'm a little confused about time zone config, is it enough if I set just my city?

Thanks for the great work!

reynico, 2020-08-30:
Hi!
The timezone is setup using your relative shift to UTC. i.e. for Argentina is -3.

96611229, 2020-08-30:
Thank you for such a fast response! Excellent, good to know.

96611229, 2020-08-31:
Hi, I hope you don't mind me responding to this thread about another question. Didn't want to spam you with new ticket :) 

I ran the script as root, was that a good call? As root I first did "cd $HOME" and then installed the script. This bit concerns me, running crontab -l shows:

> 1 0 * * * /home/pi/raspberry-noaa/schedule.sh

If I do nano or edit that path, it opens a new file. Meaning, it cannot find it. So I am not sure if it will work. The full path appears to be at:

~/raspberry-noaa/schedule.sh 

So did I make a mistake by running as root? Is this script supposed to be installed from pi user?

Thank you!

reynico, 2020-08-31:
Hi!
The script is intended to be run as `pi` user

96611229, 2020-08-31:
Thanks! I realized that and installed it... however no files show in the browser under IP of the device. I ran it whole day and the following folders are empty:

Index of /image/2020/08/31/
Index of /image/2020/08/31/

Is there a log I can see failures in? I did some testing and I saw:

- when adding time zone, I couldn't do +2 or 2+, instead it only accepted "2" as a time zone. Is that correct? I'm in UTC+2.
- when testing "./test_reception.sh 90.3" it first didn't work, it would be "Canceled by user" because it was missing mp3 sox or something like that. After a little googling I fixed that dependency and installed it.
- I'm running the following raspios 2020-08-20-raspios-buster-armhf-lite - is this ok?

Thank you!

reynico, 2020-08-31:
Check the file `sun.py`, your timezone must be inverted. If you are in +2 then timezone should be `timezone = -2`

96611229, 2020-09-01:
Just checked and it does show `timezone = -2` however it only shows it when I enter `2` and not as `+2`. If I enter 2 it will say synthax error. Not a big deal. In the meantime I reinstalled completely and I think there may be an issue when running as pi user and not root.

I ran test reception command and I get:

`mkdir: cannot create directory ‘/var/www/wx/audio/’: Permission denied
mkdir: cannot create directory ‘/var/www/wx/meteor’: Permission denied
mkdir: cannot create directory ‘/var/www/wx/meteor’: Permission denied
mkdir: cannot create directory ‘/var/www/wx/image/’: Permission denied
mkdir: cannot create directory ‘/var/www/wx/image’: Permission denied`

    `The server is in testing mode tuned to 90.2 Mhz!
    Open a terminal in your computer and paste:
    ncat 202 8073 | play -t mp3 -`
    

`Found 1 device(s):
sox FAIL formats: no handler for given file type `mp3'
  0:  , d??~đ?~???~???v@P?v???v, SN: T??~pi?v`

`Using device 0: Generic RTL2832U OEM
usb_open error -3
Please fix the device permissions, e.g. by installing the udev rules file rtl-sdr.rules
Failed to open rtlsdr device #0.`

96611229, 2020-09-01:
In addition to my previous comment, running the cronjob shows a number of permission denied errors:

pi@raspberrypi:~/raspberry-noaa $ /home/pi/raspberry-noaa/schedule.sh
mkdir: cannot create directory ‘/var/www/wx/audio/’: Permission denied
mkdir: cannot create directory ‘/var/www/wx/meteor’: Permission denied
mkdir: cannot create directory ‘/var/www/wx/meteor’: Permission denied
mkdir: cannot create directory ‘/var/www/wx/image/’: Permission denied
mkdir: cannot create directory ‘/var/www/wx/image’: Permission denied
mkdir: cannot create directory ‘/var/www/wx/audio/’: Permission denied
mkdir: cannot create directory ‘/var/www/wx/meteor’: Permission denied
mkdir: cannot create directory ‘/var/www/wx/meteor’: Permission denied
mkdir: cannot create directory ‘/var/www/wx/image/’: Permission denied
mkdir: cannot create directory ‘/var/www/wx/image’: Permission denied
INFO : Pass prediction in progress
INFO : Pass prediction in progress
INFO : Pass is above 30, that is OK for me
METEOR-M2 20200901-200122 89
warning: commands will be executed using /bin/sh
job 1 at Tue Sep  1 20:01:00 2020
INFO : Pass prediction in progress
mkdir: cannot create directory ‘/var/www/wx/audio/’: Permission denied
mkdir: cannot create directory ‘/var/www/wx/meteor’: Permission denied
mkdir: cannot create directory ‘/var/www/wx/meteor’: Permission denied
mkdir: cannot create directory ‘/var/www/wx/image/’: Permission denied
mkdir: cannot create directory ‘/var/www/wx/image’: Permission denied
NOAA19 20200901-181417 83
warning: commands will be executed using /bin/sh
job 2 at Tue Sep  1 18:14:00 2020
mkdir: cannot create directory ‘/var/www/wx/audio/’: Permission denied
mkdir: cannot create directory ‘/var/www/wx/meteor’: Permission denied
mkdir: cannot create directory ‘/var/www/wx/meteor’: Permission denied
mkdir: cannot create directory ‘/var/www/wx/image/’: Permission denied
mkdir: cannot create directory ‘/var/www/wx/image’: Permission denied
NOAA18 20200901-212509 83
warning: commands will be executed using /bin/sh
job 3 at Tue Sep  1 21:25:00 2020
mkdir: cannot create directory ‘/var/www/wx/audio/’: Permission denied
mkdir: cannot create directory ‘/var/www/wx/meteor’: Permission denied
mkdir: cannot create directory ‘/var/www/wx/meteor’: Permission denied
mkdir: cannot create directory ‘/var/www/wx/image/’: Permission denied
mkdir: cannot create directory ‘/var/www/wx/image’: Permission denied
NOAA15 20200901-190120 61
warning: commands will be executed using /bin/sh
job 4 at Tue Sep  1 19:01:00 2020

reynico, 2020-09-01:
Hi!
Seems like permissions were screwed up when schedule.sh ran with root user. 
Do a

```
sudo chown -R www-data:www-data /var/www/wx/
```

to fix the permissions and try again

96611229, 2020-09-02:
That worked, thanks. I ran the schedule.sh as pi now and got the following output:

pi@raspberrypi:~ $ /home/pi/raspberry-noaa/schedule.sh
INFO : Pass prediction in progress
INFO : Pass prediction in progress
INFO : Pass prediction in progress
INFO : Pass is above 30, that is OK for me
METEOR-M2 20200902-194135 61
warning: commands will be executed using /bin/sh
job 5 at Wed Sep  2 19:41:00 2020
INFO : Pass prediction in progress
NOAA19 20200902-180235 79
warning: commands will be executed using /bin/sh
job 6 at Wed Sep  2 18:02:00 2020
NOAA18 20200902-112619 37
warning: commands will be executed using /bin/sh
job 7 at Wed Sep  2 11:26:00 2020
NOAA18 20200902-211323 66
warning: commands will be executed using /bin/sh
job 8 at Wed Sep  2 21:13:00 2020
NOAA15 20200902-183636 36
warning: commands will be executed using /bin/sh
job 9 at Wed Sep  2 18:36:00 2020

Not sure why it threw those permission denied errors the first time, when I ran it as pi (not as root at all before). This was on a clean install. Is there a specific version of raspbian, now called raspios, that you tested this script agains?

Anyway, it scheduled jobs and I'll wait to see the results. 

Thank you for your help!

reynico, 2020-09-06:
I'm using Raspbian on my nodes

```
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster
```

reynico, 2020-09-26:
Closing by inactivity

closed 2020-09-26 by reynico