reynico/raspberry-noaa #38: zero length images?

surfrod, 2020-08-31:
**Describe the bug**
Images not getting created properly

**Screenshots and/or logs**

NOAA1520200830-185447-MCIR-precip.jpg              30-Aug-2020 22:54                   0

while earlier in the night
NOAA1820200830-223541-MCIR-precip.jpg              31-Aug-2020 02:51              421270

seems to have plenty of space
pi@raspberrypi:~/raspberry-noaa $ df
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root        7313744 5853292   1115044  84% /
devtmpfs         1966356       0   1966356   0% /dev
tmpfs            2000148       0   2000148   0% /dev/shm
tmpfs            2000148   33304   1966844   2% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs            2000148       0   2000148   0% /sys/fs/cgroup
/dev/mmcblk0p1    258096   39979    218117  16% /boot
tmpfs             400028      12    400016   1% /run/user/1000
tmpfs             102400       0    102400   0% /var/ramfs

**Server information (please complete the following information):**
 - OS: [e.g. Raspbian]
 - Python version [e.g. 3.7.3]
 - SDR device [e.g. RTL-SDR v3]

**Additional context**
Add any other context about the problem here.

Need info on how to turn on debugging and wondered about intermediate files etc.

I had stored some wav files and commented the rm *.wav (not exact line but you know which one) to 
save and use the audio in another decoder, but I since added the comment back, and deleted
all the extraneous wav files from previous passes (which weren't too many but just to be sure).

reynico, 2020-08-31:
Hi!
use the `mail` command to see your mail inside the Raspberry PI. The run logs are stored in those mails.

surfrod, 2020-08-31:
Ok... been over 15 years since I actually programmed in *nix. And I realize this is not a programming question but still...

I don’t have mail on my pi

PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games

Pi@raspberrypi:~ $ mail
bash: mail: command not found

So, what’s the best path (no pun intended) forward?

reynico, 2020-08-31:
woosh! Mine's came installed at `/usr/bin/mail`. Anyways I'll add some logging to disk stuff as it gets harder during debugging. mail is part of mailutils so doing `sudo apt install mailutils mailutils-common` should suffice to debug a bit for now.

surfrod, 2020-09-01:
Thanks... that killed my system :). At least in the installation I lost my.vnc connection and, being headless, need to fix this to continue. More tomorrow, thanks for the help!

surfrod, 2020-09-01:
the pi is back up and “seems” to be working, though I saw the Meteor 2 processing ran out of space?? Still have to investigate that, possibly maybe tmp space since that is a smaller mount point?  Saw some really funky unprocessed files from different sats last night but as far as I could tell, it’s something in the telemetry and not anything to do with my pi, per se.

There is a permission problem with a default installation where the /var/www tree did now have write permissions for the user: pi, which is the default Cron user.

surfrod, 2020-09-01:
At any rate, I think the original issue is “closed” unless you still want to track related comments here.

Thanks for the help and pointer w.r.t. mail ++

reynico, 2020-09-01:
Hi! Sorry about the system screw-up!

> though I saw the Meteor 2 processing ran out of space??
Hmm, that's weird as it does not take too much space. There's nothing to do with the ramfs space as it's only used for noaa reception.

> There is a permission problem with a default installation where the /var/www tree did now have write permissions for the user:
Could you please post the result of `groups`? I'm expecting something like this where pi is part of the www-data group
```
pi@wx:~ $ groups
pi adm dialout cdrom sudo audio www-data video plugdev games users input netdev gpio i2c spi
````

surfrod, 2020-09-02:
Here is the error processing Meteor audio:

+ echo 'INFO : Normalization in progress'
INFO : Normalization in progress
+ sox /var/www/wx/meteor/audio/METEOR-M220200902-090357.wav /var/www/wx/meteor/M
ETEOR-M220200902-090357.wav gain -n
sox FAIL gain: error writing temporary file: No space left on device
sox FAIL sox: `/var/www/wx/meteor/METEOR-M220200902-090357.wav' error writing ou
tput file: No space left on device

Any idea which partician may be causing the problem, or possibly this is some other error “masked” as a space error?

pi@raspberrypi:~/raspberry-noaa $ df
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root        7313744 5977452    990884  86% /
devtmpfs         1966356       0   1966356   0% /dev
tmpfs            2000148       0   2000148   0% /dev/shm
tmpfs            2000148   16920   1983228   1% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs            2000148       0   2000148   0% /sys/fs/cgroup
tmpfs             102400       0    102400   0% /var/ramfs
/dev/mmcblk0p1    258096   39979    218117  16% /boot
tmpfs             400028       4    400024   1% /run/user/1000

reynico, 2020-09-06:
Uhm, that's weird. That's happening only for meteor files, right?

surfrod, 2020-09-06:
Yes, only Meteor files.

reynico, 2020-09-13:
Hi!
I couldn't replicate the behaviour you see there, so I can only recommend a reboot of the Raspberry PI to clean up the temporary storage (such as /tmp and ramfs) and see if that helps

surfrod, 2020-09-13:
Thanks, I have rebooted many times. I will have to put more debugging in. I was asking 3 other questions ( not related to this) that all " took care of themselves " 👍 
.... but this one hasn't

surfrod, 2020-10-07:
Space solved the issue. Upgraded to a 64gig sd card and all is right with the world (except for a couple of other open issues ... unrelated to space :) )

surfrod, 2020-10-13:
sorry to reopen, but I'm seeing

Output at 60000 Hz.
/usr/bin/sox FAIL sox: `/var/ramfs/audio/NOAA1520201013-090704.wav' error writin
g output file: No space left on device
Signal caught, exiting!

and df -h shows:

pi@raspberrypi:~ $ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        59G  9.3G   47G  17% /
devtmpfs        1.9G     0  1.9G   0% /dev
tmpfs           1.9G  4.0K  1.9G   1% /dev/shm
tmpfs           1.9G  8.6M  1.9G   1% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           1.9G     0  1.9G   0% /sys/fs/cgroup
tmpfs          1000M     0 1000M   0% /var/ramfs
/dev/mmcblk0p1  253M   54M  199M  22% /boot
tmpfs           383M  4.0K  383M   1% /run/user/1000
/dev/sda1        60G   16G   45G  26% /media/pi/USB20FD

I'm totally confused as I am seeing this repeatedly now!

surfrod, 2020-10-16:
seems like there is some discrepancy in receive_meteor.sh that is leaving .wav files in /var/ramfs

INFO : Normalization in progress
+ sox /var/ramfs/audio/METEOR-M220201016-092944.wav /var/www/wx/meteor/METEOR-M220201016-092944.wav gain -n
sox WARN dither: dither clipped 1 samples; decrease volume?
+ log 'Demodulation in progress (QPSK)' INFO
+ local 'log_message=Demodulation in progress (QPSK)'
+ local log_priority=INFO
+ [[ -n 1 ]]
+ ((  1 < 0  ))
+ echo 'INFO : Demodulation in progress (QPSK)'
INFO : Demodulation in progress (QPSK)
+ meteor_demod -B -o /var/www/wx/meteor/METEOR-M220201016-092944.qpsk /var/www/wx/meteor/METEOR-M220201016-092944.wav

Meteor-M2 LRPT demodulator v0.3-beta1

(09:46:18) Input: /var/www/wx/meteor/METEOR-M220201016-092944.wav, output: /var/
www/wx/meteor/METEOR-M220201016-092944.qpsk
+ DELETE_AUDIO=true
+ '[' true = true ']'
+ log 'Deleting audio files' INFO
+ local 'log_message=Deleting audio files'
+ local log_priority=INFO
+ [[ -n 1 ]]
+ ((  1 < 0  ))
+ echo 'INFO : Deleting audio files'
INFO : Deleting audio files
+ rm /var/www/wx/meteor/audio/METEOR-M220201016-092944.wav
rm: cannot remove '/var/www/wx/meteor/audio/METEOR-M220201016-092944.wav': No su
ch file or directory
+ rm /var/ramfs/METEOR-M220201016-092944.wav
rm: cannot remove '/var/ramfs/METEOR-M220201016-092944.wav': No such file or dir
ectory
+ log 'Decoding in progress (QPSK to BMP)' INFO
+ local 'log_message=Decoding in progress (QPSK to BMP)'
+ local log_priority=INFO

closed 2020-10-16 by surfrod