reynico/raspberry-noaa #39: Using the software with a Nooelec Sawbird Noaa

tomamdy, 2020-09-06:
I am very happy with the software and receive Noaa pictures well without a Sawbird. But I try tu use a Nooelec Sawbird Noaa and the I get no pictures. Only grey noise. I have tried different gain settings, with no results. What I an doing wrong ?

reynico, 2020-09-06:
Hi! I've added a new test reception script: https://github.com/reynico/raspberry-noaa/blob/master/test_reception.sh

so you can test if the receiver is working properly by listening to a near broadcast fm

tomamdy, 2020-09-06:
But the sawbird is for 136 - 137 Mhz and I think it is another story. Without Sawbird receiving is good, but I thought could be better with the Sawbird Noaa by Nooelec.

reynico, 2020-09-13:
Hi!
Sadly I couldn't get any Sawbird noaa to test. Is there any chance to QRP transmit over the 136mhz band to test reception using the test_reception.sh script?

tomamdy, 2020-09-13:
Thank You for your kind reply. Receiving works ok, but the only problem is the gain wih the Sawbird. I yhought, may be another one reads this problem. I know it is a little bit off topic.

closed 2020-09-16 by tomamdy