reynico/raspberry-noaa #40: Not an issue: Twitter Develepor account rejected

saudfs, 2020-09-15:
First I would like to thank you for your creation, you really helped in making amateur satellites become much more accessible and convenient. 

now on to my problem, I applied for a Twitter Developer account but I was rejected, don't know why. Is there an alternative way to automatically tweet my images? I don't know where else to ask, and since other people might face this problem it might be helpful to find an alternative to Tweepy.

reynico, 2020-09-15:
Hi! Thanks for the kind words

I'll investigate about it, sadly this is not a limitation of Tweepy but Twitter. Do the said why your request was denied?

surfrod, 2020-09-15:
This happened to me but you just have to respond to their inquiries. Specifically I found in their rules that it specifically allowed for posting of weather images. In my response I quoted that rule and then stated I was using public domain images downloaded from NOAA weather sats. 

I received the approval shortly after that.

Good luck!

saudfs, 2020-09-15:
> Hi! Thanks for the kind words
> 
> I'll investigate about it, sadly this is not a limitation of Tweepy but Twitter. Do the said why your request was denied?

You are right, it's a Twitter issue not Tweepy. They did not explain anything in the email they sent and this is the email I received:

"Your Twitter developer account application was not approved.

Thank you for your interest in developer access. We are unable to serve your use case at this time.

Thank you for your interest in building on Twitter."

Really annoying! That's why I'm trying find an alternative to Tweepy.

@surfrod Did they actually ask you or did you contact them after they denied your request? Because I only got the email and they also mention that there's currently no way to appeal the decision.

surfrod, 2020-09-15:
just respond back and state your desire to post public domain NOAA satellite weather images, with an approximate rate of less than 20 per day.

I did receive several emails, but I don't recall if they sent the links to the rules in the first, or after my first reponse.

surfrod, 2020-09-15:
I received a review email then they asked for this:

Before we can complete our review of your developer account application, we need some more details about your use case, and who will be using your product. 

Please reply to this email with the following information:

    A list of the government or public sector entities that will have access to Twitter content, or information derived from Twitter content, under this use case.
    The specific use cases of your product or service by government or public sector entities.

Once we’ve received your response, we’ll continue our review. We appreciate your help!
----

after my response I was told I was denied similar to your email then I responded back asking for clarification and I gave the info I stated above. Might have taken 2 responses but I did finally get approved

saudfs, 2020-09-15:
Did you reply from your email or from the twitter developer website? I didn't get a follow up for my application, they just denied it 2 hours after I applied.

surfrod, 2020-09-15:
I replied from the email.

You may want to reapply, and carefully answer the questions. My reply was in the form of a "case number" to which I replied.

1st reply to the application was: Verify your Twitter Developer Account
2nd was a case# twitter develop account application email stating they are reviewing
3rd email was a request for clarification about my application:
"

Before we can complete our review of your developer account application, we need some more details about your use case, and who will be using your product.

Please reply to this email with the following information:

  *   A list of the government or public sector entities that will have access to Twitter content, or information derived from Twitter content, under this use case.
  *   The specific use cases of your product or service by government or public sector entities.

Once we’ve received your response, we’ll continue our review. We appreciate your help!

"

I replied and was then denied, to which I clarified a couple of more times including the quote from their rules.

Always check your spam folder to see if anything is in there as well (in general)

saudfs, 2020-09-15:
This is what I see when I try to apply again from the same account:

Thank you for your interest in the Twitter developer platform. Based on our review of your use case, we are unable to approve your developer application at this time.

   
- Applications may be rejected if they are found to be in violation of any section of the Developer Agreement and Policy, Automation Rules, Display Requirements, and/or the Twitter Rules.

- We don’t currently allow you to appeal this decision. We are investigating options to allow people who feel they’ve been inappropriately rejected to appeal. Please stay informed for future updates.

- We cannot comment on specific applications in public channels, including through official Twitter handles or our developer forum.

I'll reply to the rejection email and see what happens, maybe they will change their minds.

saudfs, 2020-09-15:
@surfrod just to be sure, what type of account did you apply for? I applied for a bot account.

surfrod, 2020-09-15:
Not sure but I think it was a developer account

saudfs, 2020-09-25:
Sorry for being late reporting back. 
 Right now it's impossible to reapply for a developer account on Twitter, I went to the Twitter developer community at twittercommunity.com and many had the same problem, they all received the same TLDR reply: there is no way to appeal the decision, we might change it in the future.

closed 2020-10-23 by reynico