reynico/raspberry-noaa #42: [Error]

Indr1d-C0ld, 2020-09-21:
Hello, thanks for this great project I'm currently using with success, after a little customization - just about this, can you help me adding 2 more enhancements to the results? I would like to post process in 'thermal' and 'sea thermal' if possibile, I've opened and tried to edit receive.sh file but I'm afraid I can do mess here...

Thanks for your work.

reynico, 2020-09-22:
Hi!
Just add the enhancements here: https://github.com/reynico/raspberry-noaa/blob/master/receive.sh#L36

As
```
ENHANCEMENTS="ZA MCIR MCIR-precip MSA MSA-precip HVC-precip HVCT-precip HVC HVCT SEA THERM"
```

Indr1d-C0ld, 2020-09-22:
Thanks Nico, after 'else'? Only if the Sun has a certain elevation?

reynico, 2020-09-22:
You may need to test that case a bit. I'm not really sure if SEA and THERM works without visible light

Indr1d-C0ld, 2020-09-23:
Thanks again - strange, it seems I don't get any of these enhanced images, day or nightime, after modified the file: maybe there's a maximum of 4 pictures uploaded to Twitter?

reynico, 2020-09-23:
I don't recall the Twitter limit number but yes, it's 4'ish.

Anyways the pictures with all enhancements are stored in your Raspberry PI so you can access them anytime through the browser

Indr1d-C0ld, 2020-09-23:
Right, didn't think about that - I'll check when I'll be back home this evening, thanks Nico.

Indr1d-C0ld, 2020-09-25:
Ok Nico, new enhancements are perfectly stored in web server - I've had an interesting reading about the meaning behind each of the NOAA enhancements, and I would like to post on Twitter only MCIR + HVCT-PRECIP + SEA + THERMAL: how can I mod receive.sh or post.sh to do that? Thanks for your help

reynico, 2020-09-26:
Hi!
Replace the filenames in this line: 
https://github.com/reynico/raspberry-noaa/blob/master/receive.sh#L51
```
		python3 "${NOAA_HOME}/post.py" "$1 ${START_DATE}" "$7" "${NOAA_OUTPUT}/image/${FOLDER_DATE}/$3-MCIR-precip.jpg" "${NOAA_OUTPUT}/image/${FOLDER_DATE}/$3-MSA-precip.jpg" "${NOAA_OUTPUT}/image/${FOLDER_DATE}/$3-HVC-precip.jpg" "${NOAA_OUTPUT}/image/${FOLDER_DATE}/$3-HVCT-precip.jpg" 
```

with the desired enhancements

Indr1d-C0ld, 2020-09-27:
Thanks friend, it perfectly worked - I can confirm SEA and THERM enhancements work in the nighttime too.

reynico, 2020-09-27:
Good to hear!

closed 2020-09-27 by reynico