reynico/raspberry-noaa #43: Everytime I get this error

tomamdy, 2020-09-24:
/home/pi/.noaa.conf: regel 9: 52.70333: opdracht niet gevonden

reynico, 2020-09-26:
Hi Tom, have you sorted out the problem?

tomamdy, 2020-09-27:
Nici,
Is this wrong ?

NOAA_HOME=/home/pi/raspberry-noaa
NOAA_OUTPUT=/var/www/wx
METEOR_OUTPUT=/var/www/wx/meteor
NOAA_AUDIO=/var/ramfs
SAT_MIN_ELEV=30
METEOR_MIN_ELEV=30
SUN_MIN_ELEV=10
LOG_LEVEL=DEBUG
LAT=52.7166667
LON=5.2833333
BIAS_TEE="enable_bias_tee"
DELETE_AUDIO=true

reynico, 2020-09-27:
That looks good! I'm in doubt about the amount of decimal positions of LAT and LON. I'm just using 4 decimal places (i.e. 52.7166 instead of 52.7166667)

tomamdy, 2020-09-27:
Thank You, I shall give it a try. A bird hit my aerial V-dipole. Stupid bird. May be thats the reason of my problems.

reynico, 2020-10-16:
Hi @tomamdy have you had a chance to fix your antenna?

tomamdy, 2020-10-16:
I have not yet fixed my antenna,

reynico, 2020-10-16:
Sad to hear! I'll close this issue for now, feel free to reopen it

closed 2020-10-16 by reynico