reynico/raspberry-noaa #44: Not an Issue: METEOR-M M2 IR channel activated

saudfs, 2020-09-30:
Hi @reynico , I don;t know if you heard but I have seen multiple accounts on twitter posting photos from METEOR-M M2 at night because of the activation of the IR channel. Is there anything we need to do to receive these transmissions?

reynico, 2020-10-02:
Hi!
I'll push an update to decode the APID 68

saudfs, 2020-10-02:
Thanks @reynico , when you're done can you please explain what do we need to do? Will I need to redo the whole installation or something else? please forgive my noobness.

reynico, 2020-10-03:
Hi @saudfs just update your master copy

```
cd ~/raspberry-noaa
git fetch && git pull
```

Recent changes includes a fix on rectify (I've also open a PR on upstream https://github.com/dbdexter-dev/meteor_rectify/pull/4) and enable receive_meteor.sh to parse the .dec file just on APID 68 (IR Channel)

Indr1d-C0ld, 2020-10-03:
Fantastic, thanks a lot Nico, I'll update in a few seconds.

Indr1d-C0ld, 2020-10-03:
Nico, I've just posted Meteor image on Twitter (night time here in Italy) but the result is black.

saudfs, 2020-10-03:
Thanks @reynico , I'll do that now.

reynico, 2020-10-03:
Hi @Indr1d-C0ld ! could you please upload the .dec file somewhere so I can check over it?

Indr1d-C0ld, 2020-10-03:
Sure my friend:

https://drive.google.com/file/d/1TMD-CjrmgiIbBXzqpk2K0orIw75RFuWX/view?usp=sharing

reynico, 2020-10-03:
Yes, seems like the bmp to jpg conversion brokes something important at the image file (maybe screwed up headers?)

I pushed a last time commit to fix that: https://github.com/reynico/raspberry-noaa/commit/c842323a22c4f89d4b95fc805df75ca0b115d465

It just feeds rectify with the raw BMP file instead of converting to JPG then moving to rectify

Indr1d-C0ld, 2020-10-03:
Perfect, awesome work and fast too... keep in touch dear dev, in the meanwhile I've modified those 2 lines in receive_meteor.sh

reynico, 2020-10-07:
Great! I'll close this issue for now, feel free to re open it if needed

closed 2020-10-07 by reynico