reynico/raspberry-noaa #45: Error after schedule.sh execution

Indr1d-C0ld, 2020-10-03:
Hello, after the update every time schedule.sh is executed I see this error:

schedule_iss.sh: 8: /home/pi/raspberry-noaa/common.sh: Syntax error: "(" unexpected
INFO : Pass prediction in progress

(Raspbian Buster + RTL-SDR v3 + Python 2.7.16)

reynico, 2020-10-03:
Hi!
Have you uncommented the ISS schedule lines from schedule.sh? ISS stuff is still WIP from my side, hereby those lines are commented here https://github.com/reynico/raspberry-noaa/blob/master/schedule.sh

Indr1d-C0ld, 2020-10-03:
Yes friend, and always used with success! It worked really well before, never seen that error - maybe it will work the same?

reynico, 2020-10-03:
ah, so I introduced a new bug then 😄

for now please just comment the schedule_iss.sh line from schedule.sh. i'll take a look later

Indr1d-C0ld, 2020-10-03:
OK, I must to tell you I've used as said before ISS reception very well, and decoded (manually, a part) lots of SSTV images from the generated audio files... awesome job Nico. I'll keep returning here for news and updates ;)

reynico, 2020-10-04:
@Indr1d-C0ld this commit fixes the problem https://github.com/reynico/raspberry-noaa/commit/128aedac2612fb166e0792596037bf1c52d3a6e3

Indr1d-C0ld, 2020-10-04:
Ok, thanks Nico - strangely still not decoded or posted anything this morning, must check if everything is ok...

Indr1d-C0ld, 2020-10-04:
I've seen /var/mail/pi doesn't update too, strange

reynico, 2020-10-07:
Hi! Could you please do a last `git fetch && git pull` so you grab the last changes? I'm afraid I introduced sort of a bug somewhere but should be working from the current master (at least on 3 of my nodes it's working fine)

Indr1d-C0ld, 2020-10-08:
Hi Nico! 

A couple of days ago I've formatted my node and configured from *clean* Buster Lite installation - before the installation I've took a look at install.sh to manage and customize a few things, and after installation the same things on various files, to implement different Twitter posts, Meteor enhancements, directory for ISS audio under web server etc.

It seems it works as scheduled, but I'm getting 'cutted' images (in the old system config the image results were larger in area, I hope to explain this well to you... now I see less, like it's missing upper part and lower part and it has been decoded only central one) - the same for Meteor, and no IR enh for it yet, only one time but completely messed up.

reynico, 2020-10-09:
Hi!
There's a `-A` parameter for wxtoimg that don't cut the low level top/bottom parts: https://github.com/reynico/raspberry-noaa/issues/16#issuecomment-654853107

anyways nothing changed about it on those updates so seems like you're losing reception somewhere (antenna, cable, connections)

Indr1d-C0ld, 2020-10-09:
Thanks dear Nico, I'll check everything after work time this evening ;)

surfrod, 2020-10-10:
I (unnecessarily) re-ran install.sh yesterday... while it didn’t complete, I notice that I am getting this error now:

mkdir: cannot create directory ‘/var/www/wx/image/2020/10/10’: Permission denied

Owner/group of /var/www/wx and decendants is www-data I believe the directory permissions are all 776.

I had changed permissions for last month to 777, but what is the long term/correct fix for this?

Does pi have to be added to a group for www-data or ?

reynico, 2020-10-10:
yeah, pi needs to be part of www-data group, this line makes it happen: https://github.com/reynico/raspberry-noaa/blob/master/install.sh#L172

saudfs, 2020-10-10:
Hi @reynico , I'm having the same issue as @surfrod after I did this:

`git checkout receive.sh`

I did that because when I tried to update with the `git fetch && git pull` I would get this error:

```
remote: Enumerating objects: 3, done.
remote: Counting objects: 100% (3/3), done.
remote: Total 3 (delta 2), reused 3 (delta 2), pack-reused 0
Unpacking objects: 100% (3/3), done.
From https://github.com/reynico/raspberry-noaa
   ab361f9..d21bcba  master     -> origin/master
Updating 6f7ce04..d21bcba
error: Your local changes to the following files would be overwritten by merge:
	receive.sh
Please commit your changes or stash them before you merge.
Aborting
```

I read around a bit until I found that solution, but I guess that created more issues for me. Is there anything I can do on my side?

reynico, 2020-10-11:
Hi! What's the content of your /home/pi/.noaa.cfg file? Seems like your file doesn't have this line

```
RAMFS_AUDIO=/var/ramfs
```

if not, add it

Indr1d-C0ld, 2020-10-11:
> Hi! What's the content of your /home/pi/.noaa.cfg file? Seems like your file doesn't have this line
> 
> ```
> RAMFS_AUDIO=/var/ramfs

NOAA.CONF I suppose, right Nico?

I'll post my situation here for the others, sorry for my replies to commit d21bcba.
By modifying .noaa.conf alone I've got 'can't write audio folder etc.' error in every schedule.sh session - I'm using RPi3B+.

I've tried to do this:

sudo umount /var/ramfs
sudo rm -rf /var/ramfs
sudo mkdir -p /var/ramfs
sudo chown -R www-data:www-data /var/ramfs/
[sudo chmod -R 775 /var/ramfs/]
sudo mount /var/ramfs/

and the error disappeared while running schedule.sh... but I got no more NOAA receptions and posts.
Since last update I've got messed up and cutted Meteor/NOAA images and then after the change in .noaa.conf no NOAA at all.

After restored previous SD image, it works again (NOAA reception too) but strangely, if I apply the fix to .noaa.conf then schedule.sh gets mad for the permission etc. so at the moment it works again but with NOAA_AUDIO lines instead of RAMFS_AUDIO.

Images now are larger and uncutted with the add of the option '-A' in receive.sh, wxtoimg line, thanks for the advice Nico.

What do you think is the problem with those changes in .noaa.conf, ramfs, etc? I've done git fetch and git pull too but same problems, starting from clean system too.

My other issue: /var/mail/pi not present and so no LOG to check, but I've discovered that the source of the problem is that I've not the postfix package in the system, but I've instead installed msmtp-mta in Buster because I've configured the Raspberry to send me notifications about hardware situation and running scripts like yours - so it's not related to your work, I must find a way to use this package for internal mail...

I hope you can help me with that strange RAMFS/NOAA_AUDIO and permissions problems, thanks.

saudfs, 2020-10-11:
Yes you're right, the line was not there. 

I added the line and did `git checkout -- receive.sh` then `git pull` . I'll report back once I record a pass and let you know if everything went well.

Thanks again!

reynico, 2020-10-11:
Hi people!

@saudfs, great! Let me know

@Indr1d-C0ld 

> I hope you can help me with that strange RAMFS/NOAA_AUDIO and permissions problems, thanks.

Let's do a sanity check here about the noaa.conf file:
```
NOAA_HOME=/home/pi/raspberry-noaa
NOAA_OUTPUT=/var/www/wx
METEOR_OUTPUT=/var/www/wx/meteor
RAMFS_AUDIO=/var/ramfs
SAT_MIN_ELEV=30
METEOR_MIN_ELEV=30
SUN_MIN_ELEV=10
LOG_LEVEL=DEBUG
LAT=-34.5549
LON=-58.4172
BIAS_TEE=""
DELETE_AUDIO="true"
```

Your config file must need this format, specially `RAMFS_AUDIO` which replaces the old `NOAA_AUDIO`

Now let's do a sanity check on the files of the project

this should not return anything:
```
cd raspberry-noaa/
grep -Ri NOAA_AUDIO *
```

 but `grep -R RAMFS_AUDIO * `  should return exactly this output: 
```
grep -R RAMFS_AUDIO * 
common.sh:if [ ! -d "${RAMFS_AUDIO}/audio/" ]; then
common.sh:	mkdir -m 775 -p "${RAMFS_AUDIO}/audio/"
receive.sh:timeout "${6}" /usr/local/bin/rtl_fm ${BIAS_TEE} -f "${2}"M -s 60k -g 50 -p 55 -E wav -E deemp -F 9 - | /usr/bin/sox -t raw -e signed -c 1 -b 16 -r 60000 - "${RAMFS_AUDIO}/audio/${3}.wav" rate 11025
receive.sh:	/usr/local/bin/wxtoimg -o -m "${NOAA_HOME}/map/${3}-map.png" -e "$i" "${RAMFS_AUDIO}/audio/${3}.wav" "${NOAA_OUTPUT}/image/${FOLDER_DATE}/${3}-$i.jpg"
receive.sh:    rm "${RAMFS_AUDIO}/audio/${3}.wav"
receive.sh:    mv "${RAMFS_AUDIO}/audio/${3}.wav" "${NOAA_OUTPUT}/audio/${3}.wav"
receive_meteor.sh:	RAMFS_AUDIO="${METEOR_OUTPUT}"
receive_meteor.sh:timeout "${6}" /usr/local/bin/rtl_fm ${BIAS_TEE} -M raw -f "${2}"M -s 288k -g 48 -p 1 | sox -t raw -r 288k -c 2 -b 16 -e s - -t wav "${RAMFS_AUDIO}/audio/${3}.wav" rate 96k
receive_meteor.sh:sox "${RAMFS_AUDIO}/audio/${3}.wav" "${METEOR_OUTPUT}/${3}.wav" gain -n
receive_meteor.sh:    rm "${RAMFS_AUDIO}/${3}.wav"
templates/noaa.conf:RAMFS_AUDIO=/var/ramfs
```

saudfs, 2020-10-11:
@reynico Good news, everything looks good for now. I just recorded a pass from NOAA 19 and everything is back to normal.

Indr1d-C0ld, 2020-10-11:
> Let's do a sanity check

Thanks for the patience Nico, I've overwritten common.sh, receive.sh and receive_meteor.sh with your latest repository files, I'm sure the problem was in those files - I've just re-done schedule.sh and no errors show up, so finger crossed... in 25min I'll have a NOAA passage, let's hope the catch will be good.

I think in the latest various updates, bug/fix and my little mods I must have lost something - I'll come back here to inform you, and thanks again Nico.

Indr1d-C0ld, 2020-10-11:
OK, the pass has been well received - I have to add again -A option to wxtoimg line, the image seems to be like 'zoomed' and so cutted. Well done mate.

reynico, 2020-10-11:
Glad to see it's working for both! Thanks for your patience on debugging/fixing this

saudfs, 2020-10-12:
Hi @reynico , my apologies for coming here again haha, but my output shows this:

```
common.sh:if [ ! -d "${RAMFS_AUDIO}/audio/" ]; then
common.sh:	mkdir -m 775 -p "${RAMFS_AUDIO}/audio/"
receive_meteor.sh:	RAMFS_AUDIO="${METEOR_OUTPUT}"
receive_meteor.sh:timeout "${6}" /usr/local/bin/rtl_fm ${BIAS_TEE} -M raw -f "${2}"M -s 288k -g 48 -p 1 | sox -t raw -r 288k -c 2 -b 16 -e s - -t wav "${RAMFS_AUDIO}/audio/${3}.wav" rate 96k
receive_meteor.sh:sox "${RAMFS_AUDIO}/audio/${3}.wav" "${METEOR_OUTPUT}/${3}.wav" gain -n
receive_meteor.sh:    rm "${RAMFS_AUDIO}/${3}.wav"
receive.sh:timeout "${6}" /usr/local/bin/rtl_fm ${BIAS_TEE} -f "${2}"M -s 60k -g 50 -p 55 -E wav -E deemp -F 9 - | /usr/bin/sox -t raw -e signed -c 1 -b 16 -r 60000 - "${RAMFS_AUDIO}/audio/${3}.wav" rate 11025
receive.sh:	/usr/local/bin/wxtoimg -o -m "${NOAA_HOME}/map/${3}-map.png" -e "$i" "${RAMFS_AUDIO}/audio/${3}.wav" "${NOAA_OUTPUT}/image/${FOLDER_DATE}/${3}-$i.jpg"
receive.sh:    rm "${RAMFS_AUDIO}/audio/${3}.wav"
receive.sh:    mv "${RAMFS_AUDIO}/audio/${3}.wav" "${NOAA_OUTPUT}/audio/${3}.wav"
templates/noaa.conf:RAMFS_AUDIO=/var/ramfs
```
Is this ok? it has everything but in different order.

And I noticed this in `mbox` for almost all the passes:

```
/usr/bin/sox FAIL sox: `/var/ramfs/audio/NOAA1820201012-093831.wav' error writin
g output file: No space left on device
```
When checking free space with `df -h` I saw this:

```
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        59G  2.5G   54G   5% /
devtmpfs        1.8G     0  1.8G   0% /dev
tmpfs           1.9G     0  1.9G   0% /dev/shm
tmpfs           1.9G  8.5M  1.9G   1% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           1.9G     0  1.9G   0% /sys/fs/cgroup
tmpfs           100M  100M     0 100% /var/ramfs
/dev/mmcblk0p1  253M   54M  199M  22% /boot
tmpfs           383M     0  383M   0% /run/user/1000
```
/var/ramfs goes back to zero when I reboot.

Then I checked noaa.conf file and noticed that `DELETE_AUDIO="true"`, could that be the cause for the No space left error?
I added the quotation marks but I'll have to wait till 1749 to see if it works, I'll report back then.

Indr1d-C0ld, 2020-10-12:
Just one more doubt, I've assigned to RPi GPU the minimum memory size, 16mb of the RPi 3B+ in use and the rest on CPU - I thought these script won't use GPU, correct?

reynico, 2020-10-16:
Hi!
@saudfs sorry about that, I fixed it on c0f5287

@Indr1d-C0ld no GPU use at all!

saudfs, 2020-10-16:
Thanks @reynico , I'll let you know if everything went well in about 4 hours.

reynico, 2020-10-16:
Hi @saudfs ! 
If everything went good feel free to close this issue

closed 2020-10-23 by reynico

saudfs, 2020-10-24:
Hi @reynico , it totally slipped my mind to let you know that everything is working great!