reynico/raspberry-noaa #46: [Error] existing rtl_fm instance running, I quit

surfrod, 2020-10-05:
**Describe the bug**
Per my previous problems with Meteor files and processing running out of space, I upgraded to a 64gig sd card.
Now I see a meteor audio file and am still searching through mail to see what happened creating the image, when
I ran across this error "ERROR : There is an existing rtl_fm instance running, I quit"

Any thoughts as to what happened, why this happened and how to get more debugging info out?

reynico, 2020-10-07:
Hi!
Time ago I introduced a condition inside receive_meteor.sh that kills any rtl_fm process to prefer the meteor passes instead of the NOAA ones.

> "ERROR : There is an existing rtl_fm instance running, I quit"

This error belongs just to receive.sh (NOAA reception) that exits the receive.sh script during a current reception (I.e. when satellite footprints are overlapping)

surfrod, 2020-10-08:
ok, so it's not an issue, just an obfuscated debug message 👍 

Thanks

closed 2020-10-08 by surfrod