reynico/raspberry-noaa #47: [Error]  levels: bad array subscript

surfrod, 2020-10-05:
**Describe the bug**

see this error in mail logs:
INFO : Starting rtl_fm record
/home/pi/raspberry-noaa/common.sh: line 17: levels: bad array subscript

the code in common.sh is:
   #check if level exists
    [[ ${levels[$log_priority]} ]] || return 1

Haven't checked anywhere but I'm not seeing log_priority set anywhere that calls common.sh.... any ideas?

reynico, 2020-10-07:
hi! Have you `git fetch && git pull` lately? I've introduced a bunch of bugs and features and fixes for those bugs so let's update your local copy to ease the debug

surfrod, 2020-10-07:
yes I did... a day or 2 ago

surfrod, 2020-10-08:
well, I was still hitting the issue, but did realize I had a brain freeze on the git fetch... I had local changes that were blocking the pull, so I moved those files into local storage and the git pull worked correctly. So, hopefully I have all your latest changes.

closed 2020-10-08 by surfrod

surfrod, 2020-10-09:
one question: in the latest versions, where is RAMFS_AUDIO supposed to be defined?

I found it undefined, but put it into .noaa.conf as RAMFS_AUDIO=/var/ramfs

reynico, 2020-10-09:
Hi! sorry about that, I forgot to update the template file. See d21bcba

surfrod, 2020-10-09:
next issue is my "ramfs" space... it's mounted on tmpfs and when processing the meteor file, that runs out of space.

strangely once I had upgraded to the 64gig sd card, that had solved any space issues until this last pull!?

current storage looks like (after removing the meteor wav file from the last pass...which caused the tmpfs use to be 100%):
pi@raspberrypi:/var/ramfs/audio $ df
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/root       61240912  7242568  51467036  13% /
devtmpfs         1926304        0   1926304   0% /dev
tmpfs            1960096        8   1960088   1% /dev/shm
tmpfs            1960096     8796   1951300   1% /run
tmpfs               5120        4      5116   1% /run/lock
tmpfs            1960096        0   1960096   0% /sys/fs/cgroup
tmpfs             102400        0    102400   0% /var/ramfs
/dev/mmcblk0p1    258096    54610    203486  22% /boot
tmpfs             392016        4    392012   1% /run/user/1000
/dev/sda1       62483488 15927584  46555904  26% /media/pi/USB20FD

For the moment, in receive_meteor.sh I pointed RAMFS_AUDIO to $NOAA_HOME/audio (newly created directory) and will see what happens.

reynico, 2020-10-09:
What's your raspberry pi model? For RPI 4 or anything with >2GiB of RAM, install.sh should create a larger ramfs partition: https://github.com/reynico/raspberry-noaa/blob/master/install.sh#L184-L188

aside from installation, receive_meteor has the same conditional for memory amount here: https://github.com/reynico/raspberry-noaa/blob/master/receive_meteor.sh#L9-L13 where if your system has less than 2GiB for RAM, RAMFS converts to METEOR_OUTPUT so you don't use the ramfs partition at all for meteor passes

surfrod, 2020-10-09:
yes, using an RPI 4

like I said, was fine until the last pull.

do I have to clean and re-install?

reynico, 2020-10-09:
No need to.

Edit the `/etc/fstab` file, look for the last line. It should be 
```
tmpfs                   /var/ramfs      tmpfs nodev,nosuid,size=1000M 0 0
```

if size is not 1000M, change it to 1000M. Then save the file and do

```
sudo mount -o remount,rw /var/ramfs
```

Now your ramfs partition should be 1GiB

```
pi@wx:~/raspberry-noaa $ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        14G  7.8G  5.6G  59% /
devtmpfs        1.8G     0  1.8G   0% /dev
tmpfs           2.0G     0  2.0G   0% /dev/shm
tmpfs           2.0G   25M  1.9G   2% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           2.0G     0  2.0G   0% /sys/fs/cgroup
tmpfs          1000M   84K 1000M   1% /var/ramfs
/dev/mmcblk0p1  253M   52M  201M  21% /boot
tmpfs           391M     0  391M   0% /run/user/1000
```

surfrod, 2020-10-09:
ok looks good now... will see how it goes!