reynico/raspberry-noaa #50: Doubt about sun.py timezone

Indr1d-C0ld, 2020-10-15:
Hey Nico, here I am (again) trying to always make better my station - in sun.py I've found wrong timezone "-2" while I'm in "+2": do I have to manually put "2" or "+2"? Maybe this is the reason of the last days bad receptions and cutted images.

reynico, 2020-10-16:
Hi! sun.py gets the timezone inverted so if you're in +2, set it to -2. Anyways this pull request https://github.com/reynico/raspberry-noaa/pull/49 fixes that. I'll merge it soon

reynico, 2020-10-16:
Hi, I've reverted back those commits to the old state. This is valid:
> sun.py gets the timezone inverted so if you're in +2, set it to -2

Indr1d-C0ld, 2020-10-16:
> Hi, I've reverted back those commits to the old state. This is valid:
> > sun.py gets the timezone inverted so if you're in +2, set it to -2
> 
> 

Ok Nico, currently Italy is in CEST so maybe do I have to put in file -1 (thinking about CEST +1) according to https://www.timeanddate.com/time/zone/italy ?

Indr1d-C0ld, 2020-10-18:
Everything is OK now Nico - it seems to be an issue with last dipole, but I've passed this night getting fixed every equipment in my radio-mast - the only strange thing at the moment is that the RPi is not posting on Twitter received and processed images, while last days it worked as always: maybe a problem of concurrent Twitter posting, I've only one station account for multiple RPis, and every one is linked to Tw and Telegram, but this is strange. In post.py I've modified in the past the text in in the post and the emoji.

Indr1d-C0ld, 2020-10-18:
UPDATE: here's the mess, that f****** new moji I've inserted in post.py:

 File "post.py", line 16
    satEmoji = u'\U0001F6F0' + u'\U000FE0F'
                              ^
SyntaxError: (unicode error) 'unicodeescape' codec can't decode bytes in position 0-8: truncated \UXXXXXXXX escape

That error didn't let the script post anymore on Twitter.

reynico, 2020-10-18:
Glad you found the problem! I'll close this issue for now

closed 2020-10-18 by reynico