reynico/raspberry-noaa #51: PPM correction for SDR @55 default

Indr1d-C0ld, 2020-10-16:
I've seen in receive.sh that the line inherent rtl_fm execution has the option for ppm calibration set to 55 - can I ask why? My SDR v3 I've found in the past by using sdr# etc. has a drift correction value of 7, for example. Can't remember, maybe the -p value for rtl_fm has to be changed in other files too ( if it has to be done)?

reynico, 2020-10-16:
holy sh. 

I'll revert the last merge. sorry about that

Indr1d-C0ld, 2020-10-16:
Holy bash!

Nico I've *always* seen those strange ppm correction values, in the past I've automatically corrected them, but I wanted you to know about this. Please check receive_iss // receive_meteor and the others receive files for rtl_fm setting line if they are right :)

reynico, 2020-10-16:
Hi @Indr1d-C0ld I reverted back the master state

closed 2020-10-18 by reynico