reynico/raspberry-noaa #52: pass wrong

drveni, 2020-10-17:
the times do not coincide with the satellite flights over my location

/home/pi/raspberry-noaa/schedule.sh or atq
example:

1st flight is
NOAA19 20201017-210158 64
warning: commands will be executed using /bin/sh
job 65 at Sat Oct 17 21:01:00 2020

im using orbirton (win)to compare flights, and the times do not match
in windows wxt (satelite pass list) is correct and work, on rasp is confused. !!? and wrong times!!?
not only are the hours wrong but the minutes are also missed..

.wxtoimgrc latitute/longitude are correct
on predict are same numbers as in .wxtoimgrc

rasp 4
Python 2.7.16
rtl-sdr v3

Indr1d-C0ld, 2020-10-17:
Predict uses inverted longitude value, if I'm right, so you've to put negative value if you're Lon E (it was documented in the install.sh step, confirmed).

reynico, 2020-10-18:
hi @drveni !
Is your Raspberry PI date set correctly? Please check both date and timezone with the `date` command.

drveni, 2020-10-19:
yes looks like times are good, and now it works when I have entered a negative sign. i get a black image as if nothing catches, i think the problem is now in the antenna location, i have to deal a little more when i catch the time. thank you very much for now.

closed 2020-10-23 by reynico