reynico/raspberry-noaa #53: predict not bug but hmm..

drveni, 2020-10-27:
im not sure if it's a bug, but I have doubts.

if I compare satellite overflight with orbicon (on win10)and with atq, from this project. some overflights are accurate, and some overflights are simply missing, from the same satellite. at first I thought the problem was that it was too far away, but the overflight was done right above my location. so I guess something with "predict" is not good!? 
as far as i can see, the data is downloaded from the same txt file server. but just fly over some missing.

next problem, can it be adjusted that some images if the satellite flies in the opposite direction are not "upside-down"? "mirror".  i know that this software should work alone to have an algorithm if the satellite is flying in the opposite direction. are there any additional settings for this? 
it is difficult to determine if they are all clouds then it is not known what is up and what is down.

i know that the images are black and white from noaa, is it possible with WXToIMG to somehow raise it to draw a grid of states and add artificial color like on a "win" system?

the meteor is sometimes a good image sometimes nothing or dashed (but the color hi-res appears and that's okay)

I apologize for the questions and doubts, I'm just wondering if something else can be set manually, I watched the scripts but I'm a total beginner..

reynico, 2020-10-27:
Hi @drveni !

> if I compare satellite overflight with orbicon (on win10)and with atq, from this project. some overflights are accurate, and some overflights are simply missing, from the same satellite. at first I [...]
as far as i can see, the data is downloaded from the same txt file server. but just fly over some missing.

If the passes atq see are good, then you need to adjust the `SAT_MIN_ELEV` and `METEOR_MIN_ELEV` settings on `/home/pi/.noaa.conf`. Those settings filter out the satellite passes below those thresholds. Using a QFH antenna on a mildly populated neighbour, 30 degrees are just OK. less than 30 degrees captures noisy passes.

--- 

> next problem, can it be adjusted that some images if the satellite flies in the opposite direction are not "upside-down"? "mirror". i know that this software should work alone to have an algorithm if the satellite is flying in the opposite direction. are there any additional settings for this?
it is difficult to determine if they are all clouds then it is not known what is up and what is down.

This is correct for NOAA passes, where wxtoimg uses the generated map to correct the image (if needed). For Meteor passes there's nothing too fancy. You may want to try setting `FLIP_METEOR_IMG` to `false`. But first let me know which passes are upside down (Meteor or Noaa, daylight or night passes?)

---

> i know that the images are black and white from noaa, is it possible with WXToIMG to somehow raise it to draw a grid of states and add artificial color like on a "win" system?

Yes, current implementation creates the false color images using the wxtoimg enhancements. Check the folder, you will find the picture names annotated with their enhancement such as HVC, HVCT, MSA, MSA-precip, etc. - those are colorised images.

---

> the meteor is sometimes a good image sometimes nothing or dashed (but the color hi-res appears and that's okay)

do you have any example I can see of this? In my humble opinion, that's pretty much a low SNR caused by local electromagnetic interference (as broadcast FM stations). Do you use any kind of filter and/or amplifier?

drveni, 2020-11-01:
hello, sorry I'm late with the answer, I'm in a hurry because of work.

looking pic, the last picture that is upside down is "NOAA1920201031-171542-MCIR.jpg", 31.10.2020. my location is "Latitude: 45 ° 33 '4.00" N
Longitude: 18 ° 41 '38.00 "E" Osijek / Croatia.
other pictures as far as I can see are ok for now, I also came across noaa15, but now I can't see at the moment. if I run into it I'll let you know

for meteor for now im not using filter or amplifier, im ordered and waiting for delivery..

for /home/pi/.noaa.conf thx for info !!

closed 2020-11-17 by reynico