reynico/raspberry-noaa #54: error in longtitude

tomamdy, 2020-10-27:
I get this error. Could you explain what that means.

/home/pi/.noaa.conf: line 9: syntax error near unexpected token `('
/home/pi/.noaa.conf: line 9: `LAT=Decimal latitude and longitude coordinates for Enkhuizen (Netherlands): 52.7033'

reynico, 2020-10-27:
Hi @tomamdy !
Could you please paste your current `/home/pi/.noaa.conf` file contents?

tomamdy, 2020-10-28:
NOAA_HOME=/home/pi/raspberry-noaa
NOAA_OUTPUT=/var/www/wx
METEOR_OUTPUT=/var/www/wx/meteor
RAMFS_AUDIO=/var/ramfs
SAT_MIN_ELEV=30
METEOR_MIN_ELEV=30
SUN_MIN_ELEV=10
LOG_LEVEL=DEBUG
LAT=change_latitude
LON=change_longitude
BIAS_TEE="enable_bias_tee"
DELETE_AUDIO="true"
FLIP_METEOR_IMG="true"

tomamdy, 2020-10-28:
I have changed, but still got this mail:

/home/pi/.noaa.conf: regel 9: 52.70333: opdracht niet gevonden
/home/pi/.noaa.conf: regel 9: 52.70333: opdracht niet gevonden

Regards,
Tom

reynico, 2020-11-17:
Hi @tomamdy sorry for the delay. I've been trying to replicate this error but I couldn't get to it.

Let's start from the beginning

1. How's your `.noaa.conf` file looking right now?
2. What happens when you manually call `/home/pi/raspberry-noaa/schedule.sh` ?
3. Where/when does that error `/home/pi/.noaa.conf: regel 9: 52.70333: opdracht niet gevonde` happens?

tomamdy, 2020-11-18:
Today I have installed Your latest release and anothe antenna. I think it should work now. Nice webinterface,

Tom

closed 2020-11-18 by tomamdy