reynico/raspberry-noaa #58: How to change web panel language?

saudfs, 2020-11-11:
Hi @reynico  , 

I just did a clean OS install and downloaded Raspberry-NOAA. I really love what you did there!

Is there a way to change the language of the tabs? 

Also in the passes tab it seems to show me a different time zone, for example `atq` shows me this:

```
2	Wed Nov 11 21:43:00 2020 a pi
1	Wed Nov 11 19:03:00 2020 a pi
3	Wed Nov 11 18:26:00 2020 a pi
```
but the passes tab shows me this:

[image]

Thanks again for the great work!

reynico, 2020-11-13:
Hi Saud! Glad to see you liked it!

1. Language
Edit the file `/var/www/wx/Config.php`. Replace `es` by `en`  to set the website to English. If you want to contribute your mother tongue open a pull request adding the language to the `lang` folder!

2. Timezone
Edit the file `/var/www/wx/header.php` and change the line `date_default_timezone_set`. Here's a full list of TZs: https://www.php.net/manual/en/timezones.php

I'll automate this steps somewhere on the future

saudfs, 2020-11-13:
Hi @reynico I just changed the whole es.php manually to change the language lol. I also found out how to change the time zone while playing around with the files, I came here to tell macca100 how to do it but you got there first.

I don't know how GitHub works but I will try to learn it and then submit the pull request, I'm about to introduce Raspberry-NOAA to some people and it might help a little if it's in Arabic!

reynico, 2020-11-13:
That would be great! Thank you

closed 2020-11-17 by reynico