reynico/raspberry-noaa #59: Request: add watermark to images

skuske, 2020-11-12:
It would be cool if there was an option to add a text string watermark to the generated images.

Thanks for this cool project! :o)

reynico, 2020-11-13:
Hi @skuske !

Right now the images are watermarked by `annotation` as on https://github.com/reynico/raspberry-noaa/blob/master/receive_meteor.sh#L75

Is this what you're looking?

skuske, 2020-11-13:
@reynico 

Well, I would like the images to show a string like www.mywebsite.com to show where the images are coming from in case someone steals them. So, a string watermark or something. It would be perfect if the string could be set in the web fronend, alternatively during setup...

surfrod, 2020-11-13:
these are public domain images supplied by NOAA satellites (except for METEOR), so what is the watermark supposed to indicate?

Watermarks are typically used for privately owned images.

skuske, 2020-11-13:
The watermark is supposed to indicate that the images were decoded by me and my setup, and not by someone else.

surfrod, 2020-11-13:
I understand, but as far as I know decoding doesn't give you rights to the image since the source is public domain.

that said, you can add watermarks to anything you post... just saying by definition you don't "own" the image. YMMV 😄

skuske, 2020-11-13:
I do not claim any rights, I just want to show where the picture comes from. Anyway - I'll implement that on my own. Did not want to start any discussion on that.

closed 2020-11-13 by skuske