reynico/raspberry-noaa #60: /var/www/wx/images/ permission denied error

macca100, 2020-11-13:
I wasn't getting any images produced after a pass, when I investigated the logs I found a permission denied error in the log.

If I cd to /var/www/wx/images/ and run touch test.txt I got a permission denied. If I went up a directory I could create a file just fine.

I then ran the commands to change the permissions on the folders, I'm sure there is a better way to do this. I can now create files in the directories. I don't have another satalite pass until 1930 tonight, I will check then to see if it is resolved.

sudo chmod -R 775 /var/www/wx/audio/
sudo chmod -R 775 /var/www/wx/images/

**Screenshots and/or logs**

Return-path: <pi@raspberrypi>
Envelope-to: pi@raspberrypi
Delivery-date: Fri, 13 Nov 2020 08:47:03 +0800
Received: from pi by raspberrypi with local (Exim 4.92)
        (envelope-from <pi@raspberrypi>)
        id 1kdNF0-0001Tm-Tz
        for pi@raspberrypi; Fri, 13 Nov 2020 08:47:02 +0800
Subject: Output from your job       37
To: pi@raspberrypi
Message-Id: <E1kdNF0-0001Tm-Tz@raspberrypi>
From: pi@raspberrypi
Date: Fri, 13 Nov 2020 08:47:02 +0800
Status: O
X-UID: 6

INFO : Starting rtl_fm record
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000002

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain set to 49.60 dB.
Tuned to 138392500 Hz.
Oversampling input by: 32x.
Oversampling output by: 1x.
Buffer size: 4.27ms
Sampling at 1920000 S/s.
Output at 60000 Hz.
Signal caught, exiting!

User cancel, exiting...
INFO : Bulding pass map
Satellite: NOAA 18
Pass Start: 2020-11-13 00:32:03 UTC
Pass Duration: 14:44
Elevation: 32
Azimuth: 111
Direction: southbound
..INFO : Decoding image
wxtoimg: error: could not open output file /var/www/wx/images/NOAA1820201113-083202-ZA.jpg: Permission denied
convert: unable to open image `/var/www/wx/images/NOAA1820201113-083202-ZA.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/NOAA1820201113-083202-ZA.jpg' @ error/convert.c/ConvertImageCommand/3258.
convert: unable to open image `/var/www/wx/images/NOAA1820201113-083202-ZA.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/thumb/NOAA1820201113-083202-ZA.jpg' @ error/convert.c/ConvertImageCommand/3258.
INFO : Decoding image
wxtoimg: error: could not open output file /var/www/wx/images/NOAA1820201113-083202-MCIR.jpg: Permission denied
convert: unable to open image `/var/www/wx/images/NOAA1820201113-083202-MCIR.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/NOAA1820201113-083202-MCIR.jpg' @ error/convert.c/ConvertImageCommand/3258.
convert: unable to open image `/var/www/wx/images/NOAA1820201113-083202-MCIR.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/thumb/NOAA1820201113-083202-MCIR.jpg' @ error/convert.c/ConvertImageCommand/3258.
INFO : Decoding image
wxtoimg: error: could not open output file /var/www/wx/images/NOAA1820201113-083202-MCIR-precip.jpg: Permission denied
convert: unable to open image `/var/www/wx/images/NOAA1820201113-083202-MCIR-precip.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/NOAA1820201113-083202-MCIR-precip.jpg' @ error/convert.c/ConvertImageCommand/3258.
convert: unable to open image `/var/www/wx/images/NOAA1820201113-083202-MCIR-precip.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/thumb/NOAA1820201113-083202-MCIR-precip.jpg' @ error/convert.c/ConvertImageCommand/3258.
INFO : Decoding image
wxtoimg: error: could not open output file /var/www/wx/images/NOAA1820201113-083202-MSA.jpg: Permission denied
convert: unable to open image `/var/www/wx/images/NOAA1820201113-083202-MSA.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/NOAA1820201113-083202-MSA.jpg' @ error/convert.c/ConvertImageCommand/3258.
convert: unable to open image `/var/www/wx/images/NOAA1820201113-083202-MSA.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/thumb/NOAA1820201113-083202-MSA.jpg' @ error/convert.c/ConvertImageCommand/3258.
INFO : Decoding image
wxtoimg: error: could not open output file /var/www/wx/images/NOAA1820201113-083202-MSA-precip.jpg: Permission denied
convert: unable to open image `/var/www/wx/images/NOAA1820201113-083202-MSA-precip.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/NOAA1820201113-083202-MSA-precip.jpg' @ error/convert.c/ConvertImageCommand/3258.
convert: unable to open image `/var/www/wx/images/NOAA1820201113-083202-MSA-precip.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/thumb/NOAA1820201113-083202-MSA-precip.jpg' @ error/convert.c/ConvertImageCommand/3258.
INFO : Decoding image
wxtoimg: error: could not open output file /var/www/wx/images/NOAA1820201113-083202-HVC-precip.jpg: Permission denied
convert: unable to open image `/var/www/wx/images/NOAA1820201113-083202-HVC-precip.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/NOAA1820201113-083202-HVC-precip.jpg' @ error/convert.c/ConvertImageCommand/3258.
convert: unable to open image `/var/www/wx/images/NOAA1820201113-083202-HVC-precip.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/thumb/NOAA1820201113-083202-HVC-precip.jpg' @ error/convert.c/ConvertImageCommand/3258.
INFO : Decoding image
wxtoimg: error: could not open output file /var/www/wx/images/NOAA1820201113-083202-HVCT-precip.jpg: Permission denied
convert: unable to open image `/var/www/wx/images/NOAA1820201113-083202-HVCT-precip.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/NOAA1820201113-083202-HVCT-precip.jpg' @ error/convert.c/ConvertImageCommand/3258.
convert: unable to open image `/var/www/wx/images/NOAA1820201113-083202-HVCT-precip.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/thumb/NOAA1820201113-083202-HVCT-precip.jpg' @ error/convert.c/ConvertImageCommand/3258.
INFO : Decoding image
wxtoimg: error: could not open output file /var/www/wx/images/NOAA1820201113-083202-HVC.jpg: Permission denied
convert: unable to open image `/var/www/wx/images/NOAA1820201113-083202-HVC.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/NOAA1820201113-083202-HVC.jpg' @ error/convert.c/ConvertImageCommand/3258.
convert: unable to open image `/var/www/wx/images/NOAA1820201113-083202-HVC.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/thumb/NOAA1820201113-083202-HVC.jpg' @ error/convert.c/ConvertImageCommand/3258.
INFO : Decoding image
wxtoimg: error: could not open output file /var/www/wx/images/NOAA1820201113-083202-HVCT.jpg: Permission denied
convert: unable to open image `/var/www/wx/images/NOAA1820201113-083202-HVCT.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/NOAA1820201113-083202-HVCT.jpg' @ error/convert.c/ConvertImageCommand/3258.
convert: unable to open image `/var/www/wx/images/NOAA1820201113-083202-HVCT.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/thumb/NOAA1820201113-083202-HVCT.jpg' @ error/convert.c/ConvertImageCommand/3258.
INFO : Deleting audio files

**Server information (please complete the following information):**
PI 4 
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
]

**Additional context**
Add any other context about the problem here.

reynico, 2020-11-13:
Hi @macca100 ! Thanks for the report. I'll dig into it, I think it has to do with the way I do `usermod`. Is it working now after your manual chmod?

macca100, 2020-11-16:
Hi, yes the changes to the permissions fixed the issue. I forgot to do the meteor directory but I just ran the same commands on that directory and all worked.

Thanks for putting together such awesome software, made it so easy to get images from space.

reynico, 2020-12-14:
02779d0

closed 2020-12-14 by reynico