reynico/raspberry-noaa #61: Time zone setting has no affect

macca100, 2020-11-13:
**Describe the bug**
The time shown on the passes web page shows 1 hour ahead of the pass listed in the output of the command 'atq'  I have tried modifying the the timezone when running the install script but it has no effect. 

**Screenshots and/or logs**
Screenshot attached.
[image]

Server information (please complete the following information):
PI 4
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
]

Additional context
Fresh install using raspberry pi imager.
installed raspberry-noaa using auto install script.

reynico, 2020-11-13:
hi @macca100 ! I've copypaste the answer to Saud down here:

1. Language
Edit the file `/var/www/wx/Config.php`. Replace `es` by `en`  to set the website to English. If you want to contribute your mother tongue open a pull request adding the language to the `lang` folder!

2. Timezone
Edit the file `/var/www/wx/header.php` and change the line `date_default_timezone_set`. Here's a full list of TZs: https://www.php.net/manual/en/timezones.php

I'll automate this steps somewhere on the future

reynico, 2020-11-21:
I'll close this issue for now. Feel free to reopen it if needed.

closed 2020-11-21 by reynico