reynico/raspberry-noaa #62: No reception of Meteor

schelhaas, 2020-11-17:
I have a problem receiving Meteor data. NOAA works great, but I don’t get any data from the Meteor satellite, even though the software shows me the time and elevation of its arrival.
I used the install script. Is there something else I have to do to have it recorded and automatically decoded?

pbriesch, 2020-11-17:
Try this:
`sudo chmod -R 775 /var/www/wx/meteor`

schelhaas, 2020-11-17:
I tried it, but it didn’t make a change. The audio folder is still empty, even though I don’t delete audio files.

schelhaas, 2020-11-17:
I just found the files using WinSCP on my PC. They are located in the wrong folder: /var/www/we/audio. Where do I have to make changes to correct that?
[image]

schelhaas, 2020-11-17:
My config file looks like this:
[image]

reynico, 2020-11-17:
Hi @schelhaas !

1. What's your Raspberry PI model?
2. If you set `DELETE_AUDIO`, audio files will be moved to `/var/www/wx/audio` so that's expected 

Could you please upload the file `/var/mail/pi` from your Raspberry PI to google drive or something similar so I can take a look at the logs?

schelhaas, 2020-11-17:
I am using a Pi Zero W. 
I set Delete_Audio to “false” to be able to see if the data was received. 
I will check if I can upload the file and let you know.

schelhaas, 2020-11-17:
It seems that the data stream connot get locked. Maybe that's the reason.?

Here is my file: [GoogleDrive](https://drive.google.com/file/d/1zQ-lKrdSTy-HBAobKfVr8Hj53AsjsunJ/view?usp=sharing)

reynico, 2020-11-17:
Yeah, RPI Zero doesn't have enough power to decode stuff. Are those logs from a PI Zero?

> wxtoimg: error: could not open overlay map file '/home/pi/raspberry-noaa/map/NOAA1820201115-222802-map.png'
I got those errors when using a PI Zero too. Upgrading the hardware solved them

```
(19:08:57) Input: /var/www/wx/meteor/audio/METEOR-M220201116-185334.wav, output: /var/www/wx/meteor/METEOR-M220201116-185334.qpsk
(19:08:57) Input samplerate: 96000
INFO : Deleting audio files
INFO : Decoding in progress (QPSK to BMP)
Input file "/var/www/wx/meteor/METEOR-M220201116-185334.qpsk" not found!
rm: das Entfernen von '/var/www/wx/meteor/METEOR-M220201116-185334.qpsk' ist nicht möglich: Datei oder Verzeichnis nicht gefunden
ERROR : Decoding failed, either a bad pass/low SNR or a software problem

```

here seems like the audio file is written correctly but it cannot be decoded (maybe due lack of power and/or low SnR?) 

Then there's a last METEOR pass here (METEOR-M220201117-183350.wav) but satellite lock is never achieved. Perhaps a reception problem? Are you using any kind of amplifier between antenna and receiver?

schelhaas, 2020-11-17:
Most of the logs are from the Pi Zero. I have tried a Pi3 yesterday afternoon for a couple of hours but the first error you quoted is a Pi Zero error.
I am not using an amplifier as of now. I am still testing a with a simple V-dipole, but have two more profesional antennas to be set up as soon as I have time to climb up my roof. The pass "METEOR-M220201117-183350.wav" though was recorded with my Pi Zero, too.
I will start using my Pi3 now and check it again.

PS: Is there a way to get composite pictures, too?

reynico, 2020-11-17:
Cool, let me know how it goes then

Regarding the antenna/amplifier/whatever: I strongly recommend just a FM trap (88-108Mhz filter), no amplifiers. the V-dipole may work good for NOAA but Meteor decoding is a bit difficult so you'll end up looking for a better antenna as a homemade QFH.

About composite pictures: no investigation from my side yet, sorry

schelhaas, 2020-11-17:
Thanks for your help. I have a professional QFH to be installed. I just added a 137 MHz filter to see if it helps. 
I will keep you updated.

reynico, 2020-11-21:
Thanks! I'll close this issue for now. Feel free to re open it if needed

closed 2020-11-21 by reynico