reynico/raspberry-noaa #65: Issue: Time stamp on image versus file time in web based image viewer

m0nde, 2020-11-18:
This may not be an issue but a feature or my misunderstanding as I am new to this only starting two days ago.

An example of the issue

In the web based images I have a NOAA 15 image 18/11/2020 16:20:15 
The images when viewed is stamped 18/11/2020 19:20

Please can some explain 

Thanks

Nigel M0NDE
[image]
[image]

reynico, 2020-11-19:
Hi Nigel! I forgot to document the timezone stuff. see my comment here https://github.com/reynico/raspberry-noaa/issues/64#issuecomment-729610113

m0nde, 2020-11-19:
Issue sorted by edit of /var/www/wx/header.php
Thank you Nico

closed 2020-11-19 by m0nde