reynico/raspberry-noaa #66: Meteor no decoding Pi4

Eric738, 2020-11-21:
GM 

thank a lot for your wonderful work to make a new version with thumbnails . I like !! 

just a little problem to decode meteor m2

this message i took 
Rpi 4 run here. How i Can fix this problém ?

Cordialement from France
Eric

INFO : Demodulation in progress (QPSK)

Meteor-M2 LRPT demodulator v0.3-beta1

(09:24:15) Input: /var/ramfs/audio/METEOR-M220201121-090854.wav, output: /var/www/wx/meteor/METEOR-M220201121-090854.qpsk
(09:24:15) Input samplerate: 96000
(09:24:15) Demodulator initialized
(09:24:15) (  0.0%) Carrier:   +11.5 Hz, Locked: No
(09:24:20) (  2.5%) Carrier:   +60.4 Hz, Locked: No
(09:24:25) (  5.1%) Carrier: -6295.0 Hz, Locked: No
(09:24:30) (  7.6%) Carrier: -8963.5 Hz, Locked: No
(09:24:35) ( 10.1%) Carrier: -7876.2 Hz, Locked: No
(09:24:40) ( 12.7%) Carrier: -7803.9 Hz, Locked: No
(09:24:45) ( 15.2%) Carrier:  -453.1 Hz, Locked: No
(09:24:50) ( 17.8%) Carrier: +4043.4 Hz, Locked: No
(09:24:55) ( 20.3%) Carrier: +8707.8 Hz, Locked: No
(09:25:00) ( 22.8%) Carrier: +8653.2 Hz, Locked: No
(09:25:05) ( 25.4%) Carrier: +8637.5 Hz, Locked: No
(09:25:10) ( 27.9%) Carrier: +8580.8 Hz, Locked: No
(09:25:15) ( 30.5%) Carrier: +8504.1 Hz, Locked: No
(09:25:20) ( 33.0%) Carrier: +8397.3 Hz, Locked: No
(09:25:25) ( 35.5%) Carrier: +8278.5 Hz, Locked: No
(09:25:30) ( 38.1%) Carrier: +8106.2 Hz, Locked: No
(09:25:35) ( 40.6%) Carrier: +7870.0 Hz, Locked: No
(09:25:40) ( 43.1%) Carrier: +7594.9 Hz, Locked: No
(09:25:45) ( 45.6%) Carrier: +7211.4 Hz, Locked: No
(09:25:50) ( 48.2%) Carrier: +6741.7 Hz, Locked: No
(09:25:55) ( 50.7%) Carrier: +6161.9 Hz, Locked: No
(09:26:00) ( 53.3%) Carrier: +5540.7 Hz, Locked: No
(09:26:05) ( 55.7%) Carrier: +4955.8 Hz, Locked: No
(09:26:10) ( 58.3%) Carrier: +4426.1 Hz, Locked: No
(09:26:15) ( 60.8%) Carrier: +3979.6 Hz, Locked: No
(09:26:20) ( 63.3%) Carrier: +3638.7 Hz, Locked: No
(09:26:25) ( 65.9%) Carrier: +3360.4 Hz, Locked: No
(09:26:30) ( 68.4%) Carrier: +3181.7 Hz, Locked: No
(09:26:35) ( 71.0%) Carrier: +3015.3 Hz, Locked: No
(09:26:40) ( 73.5%) Carrier: +2893.4 Hz, Locked: No
(09:26:45) ( 76.0%) Carrier: +2821.0 Hz, Locked: No
(09:26:50) ( 78.6%) Carrier: +8865.4 Hz, Locked: No
(09:26:55) ( 81.1%) Carrier: +7264.3 Hz, Locked: No
(09:27:00) ( 83.6%) Carrier: +6952.6 Hz, Locked: No
(09:27:05) ( 86.2%) Carrier: +3415.2 Hz, Locked: No
(09:27:10) ( 88.7%) Carrier:  +541.3 Hz, Locked: No
(09:27:15) ( 91.3%) Carrier: +5535.8 Hz, Locked: No
(09:27:20) ( 93.8%) Carrier: +5032.0 Hz, Locked: No
(09:27:25) ( 96.3%) Carrier: +2021.0 Hz, Locked: No
(09:27:30) ( 98.9%) Carrier: +3107.6 Hz, Locked: No
(09:27:35) Decoding completed
INFO : Deleting audio files
INFO : Decoding in progress (QPSK to BMP)
Reading /var/www/wx/meteor/METEOR-M220201121-090854.qpsk...
[2K[1G pos=   10148 (  0.01%) ( 5,10148,49) sig= -258 rs=(-1,-1,-1,-1) 9166CEEC [2K[1G pos=   33365 (  0.02%) ( 0, 6833,50) sig= -263 rs=(-1,-1,-1,-1) FC38794E [2K[1G pos=   61081 (  0.04%) ( 0,11332,49) sig= -249 rs=(-1,-1,-1,-1) D038C0B3 [2K[1G pos=   80485 (  0.06%) ( 7, 3020,51) sig= -251 rs=(-1,-1,-1,-1) CE06CA1A [2K[1G pos=   98763 (  0.07%) ( 6, 1894,50) sig= -257 rs=(-1,-1,-1,-1) B79E3B11 [2K[1G pos=  121613 (  0.09%) ( 0, 6466,49) sig= -254 rs=(-1,-1,-1,-1) B7C83E29 [2K[1G pos=  147914 (  0.11%) ( 4, 9917,49) sig= -252 rs=(-1,-1,-1,-1) 79A06B79 [2K[1G pos=  165116 (  0.12%) ( 1,  818,48) sig= -256 rs=(-1,-1,-1,-1) 45F9D9A5 [2K[1G pos=  185968 (  0.13%) ( 3, 4468,49) sig= -262 rs=(-1,-1,-1,-1) 256E6478 [2K[1G pos=  211306 (  0.15%) ( 4, 8954,49) sig= -252 rs=(-1,-1,-1,-1) 3718D6E8 [2K[1G pos=  237328 (  0.17%) ( 1, 9638,50) sig= -253 rs=(-1,-1,-1,-1) 560ADE56 [2K[1G pos=  255791 (  0.18%) ( 6, 2079,50) sig= -257 rs=(-1,-1,-1,-1) D498E682 [2K[1G pos=  280929 (  0.20%) ( 3, 8754,52) sig= -257 rs=(-1,-1,-1,-1) A84EBB5B [2K[1G pos=  298196 (  0.21%) ( 5,  883,48) sig= -257 rs=(-1,-1,-1,-1) D3E5637A [2K[1G pos=  316374 (  0.23%) ( 1, 1794,51) sig= -255 rs=(-1,-1,-1,-1) 8FBD7A00 [2K[1G pos=  336152 (  0.24%) ( 0, 3394,48) sig= -255 rs=(-1,-1,-1,-1) 50A9A65B [2K[1G pos=  366335 (  0.26%) ( 0,13799,48) sig= -252 rs=(-1,-1,-1,-1) 1D266E12 [2K[1G pos=  383761 (  0.27%) ( 5, 1042,49) sig= -259 rs=(-1,-1,-1,-1) 63746ACB [2K[1G pos=  409407 (  0.29%) ( 4, 9262,50) sig= -254 rs=(-1,-1,-1,-1) 4D78446A [2K[1G pos=  433076 (  0.31%) ( 4, 7285,49) sig= -252 rs=(-1,-1,-1,-1) E1C1DA1A [2K[1G pos=  454286 (  0.32%) ( 5, 4826,50) sig= -253 rs=(-1,-1,-1,-1) F37FF730 [2K[1G pos=  480513 (  0.34%) ( 0, 9843,48) sig= -260 rs=(-1,-1,-1,-1) 8F059310 [2K[1G pos=  505221 (  0.36%) ( 5, 8324,49) sig= -264 rs=(-1,-1,-1,-1) 98C882B6 [2K[1G pos=  537782 (  0.38%) ( 7,16177,49) sig= -246 rs=(-1,-1,-1,-1) 85CCE118 [2K[1G pos=  564971 (  0.40%) ( 0,10805,49) sig= -251 rs=(-1,-1,-1,-1) 1778EEEE [2K[1G pos=  582786 (  0.42%) ( 1, 1431,48) sig= -251 rs=(-1,-1,-1,-1) 9528E9C0 [2K[1G pos=  607293 (  0.43%) ( 6, 8123,50) sig= -256 rs=(-1,-1,-1,-1) 3FA0AC74 [2K[1G pos=  626358 (  0.45%) ( 0, 2681,50) sig= -248 rs=(-1,-1,-1,-1) 2B90351B [2K[1G pos=  644889 (  0.46%) ( 3, 2147,48) sig= -255 rs=(-1,-1,-1,-1) 55826BC9 [2K[1G pos=  671518 (  0.48%) ( 1,10245,48) sig= -254 rs=(-1,-1,-1,-1) 19196B98 [2K[1G pos=  693407 (  0.49%) ( 2, 5505,49) sig= -251 rs=(-1,-1,-1,-1) C5B6DC47 [2K[1G pos=  721103 (  0.51%) ( 2,11312,51) sig= -258 rs=(-1,-1,-1,-1) 854B5C5F [2K[1G pos=  746747 (  0.53%) ( 0, 9260,48) sig= -246 rs=(-1,-1,-1,-1) 83BD5A09 [2K[1G pos=  777304 (  0.55%) ( 6,14173,50) sig= -251 rs=(-1,-1,-1,-1) 3D2CE8E1 [2K[1G pos=  800886 (  0.57%) ( 5, 7198,49) sig= -259 rs=(-1,-1,-1,-1) C6AECE1A [2K[1G pos=  823610 (  0.59%) ( 1, 6340,48) sig= -252 rs=(-1,-1,-1,-1) 49CB7396 [2K[1G pos=  842101 (  0.60%) ( 2, 2107,50) sig= -269 rs=(-1,-1,-1,-1) A5F44A38 [2K[1G pos=  864318 (  0.62%) ( 5, 5833,49) sig= -253 rs=(-1,-1,-1,-1) 97BC0448 [2K[1G pos=  887822 (  0.63%) ( 4, 7120,48) sig= -256 rs=(-1,-1,-1,-1) 08380D35 [2K[1G pos=  906905 (  0.65%) ( 2, 2699,48) sig= -253 rs=(-1,-1,-1,-1) B3E6BC4B [2K[1G pos=  923596 (  0.66%) ( 4,  307,49) sig= -250 rs=(-1,-1,-1,-1) 93312716 [2K[1G pos=  942286 (  0.67%) ( 1, 2306,48) sig= -260 rs=(-1,-1,-1,-1) 5064F94C [2K[1G pos=  970546 (  0.69%) ( 2,11876,49) sig= -254 rs=(-1,-1,-1,-1) A5FFB2A8 [2K[1G pos= 1001275 (  0.71%) ( 4,14345,50) sig= -254 rs=(-1,-1,-1,-1) 41ED0DB6 [2K[1G pos= 1032197 (  0.74%) ( 0,14538,48) sig= -253 rs=(-1,-1,-1,-1) F670C51A [2K[1G pos= 1056117 (  0.75%) ( 1, 7536,49) sig= -252 rs=(-1,-1,-1,-1) 57C5BB56 [2K[1G pos= 1082570 (  0.77%) ( 2,10069,48) sig= -258 rs=(-1,-1,-1,-1) 5434FC30 [2K[1G pos= 1105983 (  0.79%) ( 5, 7029,49) sig= -249 rs=(-1,-1,-1,-1) 72B0458A [2K[1G pos= 1123832 (  0.80%) ( 5, 1465,50) sig= -243 rs=(-1,-1,-1,-1) 627A9BD2 [2K[1G pos= 1155474 (  0.82%) ( 1,15258,49) sig= -257 rs=(-1,-1,-1,-1) 09F6E682 [2K[1G pos= 1181479 (  0.84%) ( 1, 9621,48) sig= -256 rs=(-1,-1,-1,-1) 68450F19 [2K[1G pos= 1209833 (  0.86%) ( 4,11970,49) sig= -255 rs=(-1,-1,-1,-1) 39CACC7D [2K[1G pos= 1226446 (  0.87%) ( 1,  229,48) sig= -259 rs=(-1,-1,-1,-1) 8E786ED5 [2K[1G pos= 1251544 (  0.89%) ( 4, 8714,48) sig= -252 rs=(-1,-1,-1,-1) 49C5C819 [2K[1G pos= 1282902 (  0.92%) ( 7,14974,49) sig= -260 rs=(-1,-1,-1,-1) DD5041E2 [2K[1G pos= 1311359 (  0.94%) ( 2,12073,52) sig= -259 rs=(-1,-1,-1,-1) 18954F86 [2K[1G pos= 1340704 (  0.96%) ( 3,12961,49) sig= -249 rs=(-1,-1,-1,-1) C78A5B3A [2K[1G pos= 1361514 (  0.97%) ( 2, 4426,49) sig= -256 rs=(-1,-1,-1,-1) FE36E158 [2K[1G pos= 1389913 (  0.99%) ( 2,12015,48) sig= -253 rs=(-1,-1,-1,-1) 50D6F96B [2K[1G pos= 1415248 (  1.01%) ( 5, 8951,49) sig= -261 rs=(-1,-1,-1,-1) D2FC54B2 [2K[1G pos= 1435661 (  1.02%) ( 3, 4029,49) sig= -258 rs=(-1,-1,-1,-1) 14B6376A [2K[1G pos= 1459651 (  1.04%) ( 1, 7606,49) sig= -260 rs=(-1,-1,-1,-1) 3475E255 [2K[1G pos= 1478771 (  1.05%) ( 3, 2736,49) sig= -253 rs=(-1,-1,-1,-1) AC64D9F3 [2K[1G pos= 1507042 (  1.08%) ( 2,11887,52) sig= -257 rs=(-1,-1,-1,-1) D9E5A842 [2K[1G pos= 1535795 (  1.10%) ( 2,12369,49) sig= -257 rs=(-1,-1,-1,-1) 8533E190 [2K[1G pos= 1554226 (  1.11%) ( 0, 2047,49) sig= -261 rs=(-1,-1,-1,-1) DA646A4B [2K[1G pos= 1583162 (  1.13%) ( 1,12552,49) sig= -249 rs=(-1,-1,-1,-1) B9F81F98 [2K[1G pos= 1612190 (  1.15%) ( 1,12644,51) sig= -254 rs=(-1,-1,-1,-1) FCB82CCB [2K[1G pos= 1629027 (  1.16%) ( 3,  453,50) sig= -255 rs=(-1,-1,-1,-1) ED3BDB2E [2K[1G pos= 1660949 (  1.18%) ( 7,15538,53) sig= -257 rs=(-1,-1,-1,-1) 6E76BFCB [2K[1G pos= 1 [... cut ...] 1,-1) EF2E7A8A [2K[1G pos= 4702516 (  3.35%) ( 4, 3058,49) sig= -253 rs=(-1,-1,-1,-1) BE47C21A [2K[1G pos= 4722772 (  3.37%) ( 6, 3872,50) sig= -256 rs=(-1,-1,-1,-1) 85588AFC [2K[1G pos= 4752907 (  3.39%) ( 4,13751,50) sig= -252 rs=(-1,-1,-1,-1) B.......

Total:        260.290375
Processing:   0.000000
Correlation:  159.429001
Viterbi:      90.815506
ECC:          9.929956
Remainder:    0.115901
Packets:      0 / 5759
ERROR : Decoding failed, either a bad pass/low SNR or a software problem

reynico, 2020-11-21:
Hi Eric!

Seems like your station didn't receive any data from Meteor. Either for a low Signal to Noise Ratio or because there's a problem on the Meteor but there's not much to do from the software side

Eric738, 2020-11-21:
Wow ! odd especially since the reception of noaa 15/18/19 works perfectly. What parameter could I look at , thank?

pineapplee85, 2020-11-29:
Hi all,

I too, have a raspi 4 with no decode from meteor.
I have tried everything and I know my antenna works because I can manually receive the pictures using my laptop. 
NOAA pictures come through just fine however, the Meteor is not greyed off the pass list like NOAA ones are when they're completed. Let me know if I need to add logs, I'm not sure how to but with guidance I will be happy to. 

Also, as a feature request please could we get an option of the HVC False Colour enhancement? I really like the look of that option as it is just like a camera on the NOAA satellites. 

Thanks for the great project - it is a smashing bit of kit!

Ethan

pineapplee85, 2020-11-29:
Seems like a similar issue to Eric's where it can't store the pass in the ram.

From pi@rpi4-meteor Sun Nov 29 18:36:07 2020
Return-path: <pi@rpi4-meteor>
Envelope-to: pi@rpi4-meteor
Delivery-date: Sun, 29 Nov 2020 18:36:07 +0000
Received: from pi by rpi4-meteor with local (Exim 4.92)
	(envelope-from <pi@rpi4-meteor>)
	id 1kjRYN-0000db-Po
	for pi@rpi4-meteor; Sun, 29 Nov 2020 18:36:07 +0000
Subject: Output from your job        1
To: pi@rpi4-meteor
Message-Id: <E1kjRYN-0000db-Po@rpi4-meteor>
From: pi@rpi4-meteor
Date: Sun, 29 Nov 2020 18:36:07 +0000

mkdir: cannot create directory ‘/var/www/wx/audio/’: Permission denied
mkdir: cannot create directory ‘/var/www/wx/meteor’: Permission denied
INFO : The system doesn't have enough space to store a Meteor pass on RAM
INFO : I'll flip this image pass because FLIP_METEOR_IMG is set to true
INFO : Starting rtl_fm record
sox FAIL formats: can't open output file `/var/www/wx/meteor/audio/METEOR-M220201129-183634.wav': No such file or directory
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain set to 48.00 dB.
Tuned to 137388000 Hz.
Oversampling input by: 4x.
Oversampling output by: 1x.
Buffer size: 7.11ms
Sampling at 1152000 S/s.
Output at 288000 Hz.
Signal caught, exiting!

User cancel, exiting...
INFO : Demodulation in progress (QPSK)
[FATAL]: Could not find specified file
INFO : Deleting audio files
rm: cannot remove '/var/www/wx/meteor/audio/METEOR-M220201129-183634.wav': No such file or directory
INFO : Decoding in progress (QPSK to BMP)
Input file "/var/www/wx/meteor/METEOR-M220201129-183634.qpsk" not found!
rm: cannot remove '/var/www/wx/meteor/METEOR-M220201129-183634.qpsk': No such file or directory
ERROR : Decoding failed, either a bad pass/low SNR or a software problem

pineapplee85, 2020-12-01:
Edit: I have fixed the meteor issues!
A folder in /var/www/wx was missing.

Navigate to /var/www/wx and run

sudo mkdir -m 1777 /meteor
sudo mkdir -m 1777 /meteor/audio
sudo mkdir -m 1777 /audio

This worked for me and everything except twitter is functional on my pi 4!

Eric738, 2020-12-05:
i tested soluce explain by pineapplee85 on raspi 4, without succes. Sniff :(

Eric738, 2020-12-05:
juste wanted add that these repertory existed

reynico, 2020-12-14:
Hi, I set the chmod commands on the install script on 02779d0 . If you already setup your station, just run 
```
sudo chmod -R 775 /var/www/wx/audio/
sudo chmod -R 775 /var/www/wx/images/
sudo chmod -R 775 /var/www/wx/meteor/
```

closed 2020-12-14 by reynico