reynico/raspberry-noaa #67: pi mail gives this notice.

tomamdy, 2020-11-21:
INFO : The system doesn't have enough space to store a Meteor pass on,
Could it be that there are to much pictures

reynico, 2020-11-21:
Hi Tom!

That's ok if you're running raspberry-noaa on a system that has less than 2000MB of RAM. There's a switch in the code that stores the Meteor pass in a ramfs partition to speed up things, but that's available only if there's enough memory.

tomamdy, 2020-11-21:
Sirry, I like your siftware, but I have still problems with receiving.
[image]

reynico, 2020-11-21:
There's not much to do from software side about that. Seems like there's a lot of noise near you (broadcast FM?) but this is nothing fixable from software, sorry

tomamdy, 2020-11-21:
But a friend of me has the same problem and under Windiws ni oriblem.

tomamdy, 2020-11-21:
Here he receives by windows.. http://www.enkhuizen-meteo.nl/noaa.html

schelhaas, 2020-11-21:
By the way, I have the same experience for the last 3 days. I thought that the reason is my antenna. 
Here is how it looks when I receive NOAA. 
[image]

Before reinstalling my system 3 days ago NOAA was good.

tomamdy, 2020-11-21:
NOAA19,  Elevation: 74°,21/11/2020 17:13:50 was ok again. I am no sure, but could it be that the scandisk has not ennough space, because the old images stay on the card. Should it be better that they automatic remove.

schelhaas, 2020-11-21:
I just re-installed the version I downloaded 4 days ago and it now looks much better. Has there been a change lately? I just recognized a new language 3 days ago and in that version I got bad pictures. 

This is how it looks after my “downgrade”:
[image]

tomamdy, 2020-11-21:
This was mine.
[image]

closed 2020-12-03 by tomamdy