reynico/raspberry-noaa #68: Images stopped

trinchet, 2020-11-22:
Hi - First off, thank you for this incredible tool - it has been excellent. 

I originally loaded this on a raspberry pi and entered the wrong coordinates so got some images of China.  However, after reinstall it won't load any images.  All hardware is in place, but I noticed 2 things:
1) when I edit the .noaa.conf file the default coordinates are displaying.  It does not store my ground station coordinates.
2) after stepping through the schedule.sh and common.sh scripts I see this error: 
mkdir: cannot create directory '/audio/': Permission denied
the directory is there under /var/www/wx/ and is permissioned properly to 775.  I can't understand why the script is not seeing the directory and trying to create it.  I don't see any audio files or image files in the var/www/wx folder.

Any guidance?
Again, thank you so very much for putting this together!!

RT

reynico, 2020-11-23:
Hi! Thanks for the kind words

1. there are two files where you'll find coordinates: `.wxtoimgrc` and `.nooa.conf`. Make sure both of them are updated!
2. Mmmmm, seems like you're running the scripts manually and/or your `.noaa.conf` is screwed up? Behind `/audio/` there must be the full path which is a variable stored in `.noaa.conf`. (see RAMFS_AUDIO, NOAA_OUTPUT, METEOR_OUTPUT)

Please post your .noaa.conf file!

trinchet, 2020-11-24:
Hi!  Thanks for the reply and willingness to help!  I did run the scripts manually in an effort to try to troubleshoot the issue. My .noaa.conf file is below:

NOAA_HOME=/home/pi/raspberry-noaa
NOAA_OUTPUT=/var/www/wx
METEOR_OUTPUT=/var/www/wx/meteor
NOAA_AUDIO=/var/ramfs
SAT_MIN_ELEV=30
METEOR_MIN_ELEV=30
SUN_MIN_ELEV=10
LOG_LEVEL=DEBUG
LAT=48.2477
LON=116.3071
BIAS_TEE=""
DELETE_AUDIO=true

I think these are the default values.  I had to reinstall the package and entered these coordinates into the end of the installation:  8.81N, -79.28W and that does not seem to be updating. Today, I noticed the site does show some broken thumbnails where images would be.  (see image)

[image]

But clicking on the thumbnails shows more broken thumbnails and no images. 
[image]

Finally, I plan to move this station around when I travel so may need to modify the coordinates periodically - If I move, do I just update the values in the .noaa.conf and .wxtoimgrc files, or does more need to be done?

Thank you SO MUCH for your help with this.  It's so promising to what I need.

reynico, 2020-11-24:
>  I had to reinstall the package and entered these coordinates into the end of the installation: 8.81N, -79.28W and that does not seem to be updating.

Yeah, you'll need to manually update those values in `.noaa.conf` and `.wxtoimgrc`

> Finally, I plan to move this station around when I travel so may need to modify the coordinates periodically - If I move, do I just update the values in the .noaa.conf and .wxtoimgrc files, or does more need to be done?

That's enough!

> But clicking on the thumbnails shows more broken thumbnails and no images.

Grid is generated by calculating each satellite pass, but images are broken due to lack of reception (as of an incorrect lat/lon setup). 

Just update your lat/lon on both files, run `/home/pi/raspberry-noaa/schedule.sh`  and give a try!

trinchet, 2020-11-25:
Great!  Will do
THANK YOU!!

closed 2020-12-14 by reynico