reynico/raspberry-noaa #70: [Error] Changing Webpage Language Not Successful

mihajlo2003petkovic, 2020-11-27:
Hi, I've tried to change webpage language from Spanish to English by changing Config.php file, but the webpage is still in Spanish. Any ideas?

`<?php
return (object) array(
  'base_url' => '/images/',
  'img_per_page' => 12,
  'lang' => 'en'
);
?>
`

tomamdy, 2020-11-27:
<?php return (object) array(  'base_url' => '/images/',  'img_per_page' => 12,  'lang' => 'en');?>
 
For me, that works ok. May be pressing in Your browser F5, that refreshs our browser.

Tom

mihajlo2003petkovic, 2020-11-27:
Yes, I refreshed my browser multiple times, even rebooted Raspberry Pi, but it's still in Spanish.

tomamdy, 2020-11-27:
Should I sent You my Config.php?

mihajlo2003petkovic, 2020-11-27:
Yes, thank you!

tomamdy, 2020-11-27:
[config.txt](https://github.com/reynico/raspberry-noaa/files/5608427/config.txt)

tomamdy, 2020-11-27:
Make extension php

mihajlo2003petkovic, 2020-11-27:
I copied everything, renamed it Config.php but website is still in Spanish.

[image]

reynico, 2020-11-27:
Hi!

Keep in mind that you need to edit the Config.php inside `/var/www/wx/`

mihajlo2003petkovic, 2020-11-27:
Ohhh so that was the problem! I totally forgot about `/var/www/wx/` path! Thank you!

closed 2020-11-27 by mihajlo2003petkovic