reynico/raspberry-noaa #73: [Error] receive_meteor.sh script not working

mihajlo2003petkovic, 2020-11-30:
**Describe the bug**
A clear and concise description of what the bug is.

**Screenshots and/or logs**
If applicable, add screenshots to help explain your problem. Logs are always welcome as well

**Server information (please complete the following information):**
 - OS: [e.g. Raspbian]
 - Python version [e.g. 3.7.3]
 - SDR device [e.g. RTL-SDR v3]

**Additional context**
Add any other context about the problem here.

mihajlo2003petkovic, 2020-11-30:
I was also confused when I tried opening files:
```
## import common lib
. "$HOME/.noaa.conf"
. "$HOME/.tweepy.conf"
. "$NOAA_HOME/common.sh"
```
Everything exists except for common.sh

mihajlo2003petkovic, 2020-12-04:
```
sox FAIL formats: can't open output file `/var/www/wx/meteor/audio/METEOR-M220201204-080938.wav': Permission denied
Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain set to 7.70 dB.
Tuner error set to 55 ppm.
Tuned to 137388000 Hz.
Oversampling input by: 4x.
Oversampling output by: 1x.
Buffer size: 7.11ms
Sampling at 1152000 S/s.
Output at 288000 Hz.
Signal caught, exiting!

User cancel, exiting...
INFO : Demodulation in progress (QPSK)
[FATAL]: Could not find specified file
INFO : Deleting audio files
rm: cannot remove '/var/www/wx/meteor/audio/METEOR-M220201204-080938.wav': No such file or directory
INFO : Decoding in progress (QPSK to BMP)
Input file "/var/www/wx/meteor/METEOR-M220201204-080938.qpsk" not found!
rm: cannot remove '/var/www/wx/meteor/METEOR-M220201204-080938.qpsk': No such file or directory
ERROR : Decoding failed, either a bad pass/low SNR or a software problem
```

tomamdy, 2020-12-14:
Is there no soluyion for this faillure?

reynico, 2020-12-14:
hi! I've updated the install script. Anyways if your station is already setup, just run 
```
sudo chmod -R 775 /var/www/wx/audio/
sudo chmod -R 775 /var/www/wx/images/
sudo chmod -R 775 /var/www/wx/meteor/
```

Sorry for the delay.

closed 2020-12-14 by reynico