reynico/raspberry-noaa #78: Meteor M2 receiving

tomamdy, 2020-12-11:
Any idea why Meteor-M2 is not receving and the Noaa's are well.

pineapplee85, 2020-12-11:
Have you been receiving them before or is this a new installation?

tomamdy, 2020-12-11:
Tjis is a new installation.

tomamdy, 2020-12-12:
i@reynicotest.fritz.box Sat Dec 12 07:48:01 2020
Return-path: <pi@reynicotest.fritz.box>
Envelope-to: pi@reynicotest.fritz.box
Delivery-date: Sat, 12 Dec 2020 07:48:01 +0000
Received: from pi by reynicotest.fritz.box with local (Exim 4.92)
	(envelope-from <pi@reynicotest.fritz.box>)
	id 1knzdJ-0003vN-8U
	for pi@reynicotest.fritz.box; Sat, 12 Dec 2020 07:48:01 +0000
Subject: Output from your job       21
To: pi@reynicotest.fritz.box
Message-Id: <E1knzdJ-0003vN-8U@reynicotest.fritz.box>
From: pi@reynicotest.fritz.box
Date: Sat, 12 Dec 2020 07:48:01 +0000

mkdir: kan map ‘/var/www/wx/meteor’ niet aanmaken: Toegang geweigerd
INFO : The system doesn't have enough space to store a Meteor pass on RAM
INFO : I'll flip this image pass because FLIP_METEOR_IMG is set to true
INFO : Starting rtl_fm record
sox FAIL formats: can't open output file `/var/www/wx/meteor/audio/METEOR-M220201212-074854.wav': No such file or directory
Found 1 device(s):
  0:  Realtek, RTL2832U, SN: 00001000

Using device 0: Generic RTL2832U
usb_claim_interface error -6
Failed to open rtlsdr device #0.
INFO : Demodulation in progress (QPSK)
[FATAL]: Could not find specified file
INFO : Deleting audio files
rm: kan '/var/www/wx/meteor/audio/METEOR-M220201212-074854.wav' niet verwijderen: Bestand of map bestaat niet
INFO : Decoding in progress (QPSK to BMP)
Input file "/var/www/wx/meteor/METEOR-M220201212-074854.qpsk" not found!
rm: kan '/var/www/wx/meteor/METEOR-M220201212-074854.qpsk' niet verwijderen: Bestand of map bestaat niet
ERROR : Decoding failed, either a bad pass/low SNR or a software problem

tomamdy, 2020-12-14:
Is there no solution for this faillure?

reynico, 2020-12-14:
Hi Tom! Sorry for the delay.

I just fixed the issue on the install script. If you already setup your station just run

```
sudo chmod -R 775 /var/www/wx/audio/
sudo chmod -R 775 /var/www/wx/images/
sudo chmod -R 775 /var/www/wx/meteor/
```

closed 2020-12-14 by reynico

tomamdy, 2020-12-15:
mkdir: kan map ‘/var/www/wx/meteor/audio/’ niet aanmaken: Toegang geweigerd
INFO : Starting rtl_fm record
Found 1 device(s):
  0:  Realtek, RTL2832U, SN: 00001000

Nico: Still this problem.