reynico/raspberry-noaa #79: mkdir: kan map ‘/var/www/wx/meteor/audio/’ niet aanmaken: Toegang geweigerd

tomamdy, 2020-12-15:
mkdir: kan map ‘/var/www/wx/meteor/audio/’ niet aanmaken: Toegang geweigerd
INFO : Starting rtl_fm record
Found 1 device(s):
  0:  Realtek, RTL2832U, SN: 00001000

Nico: Still this problem.

_Originally posted by @tomamdy in https://github.com/reynico/raspberry-noaa/issues/78#issuecomment-745193231_

reynico, 2020-12-15:
Hi!
Could you please test doing 
```
chmod -R 775 /var/www/wx/meteor/audio/
```
?

tomamdy, 2020-12-15:
And then iI get this.
pi@reynicotest:~ $ mkdir: kan map ‘/var/www/wx/meteor/audio/’ niet aanmaken: Toegang geweigerd
bash: mkdir:: opdracht niet gevonden
pi@reynicotest:~ $ INFO : Starting rtl_fm record
bash: INFO: opdracht niet gevonden
pi@reynicotest:~ $ Found 1 device(s):
bash: syntaxfout nabij onverwacht symbool '('
pi@reynicotest:~ $   0:  Realtek, RTL2832U, SN: 00001000

tomamdy, 2020-12-17:
Could it be in the priveleges?
I have installed the s.w again and the same problem.

Tom

reynico, 2020-12-17:
Show me the output of
```
ls -al /var/www/wx/
```

and 
```
ls -al /var/www/wx/meteor/
```

tomamdy, 2020-12-17:
drwxrwxr-x 2 root     root      4096 Dec 17 12:51 audio
-rw-r--r-- 1 pi       pi         104 Dec 17 12:38 Config.php
drwxr-xr-x 2 pi       pi        4096 Dec 17 11:54 Controller
-rw-r--r-- 1 pi       pi         206 Dec 17 11:54 detail.php
-rw-r--r-- 1 pi       pi         146 Dec 17 11:54 footer.php
-rw-r--r-- 1 pi       pi         962 Dec 17 12:44 header.php
drwxrwxr-x 3 www-data www-data  4096 Dec 17 12:09 images
-rw-r--r-- 1 pi       pi         185 Dec 17 11:54 index.php
drwxr-xr-x 2 pi       pi        4096 Dec 17 11:54 language
-rwxr-xr-x 1 pi       pi       16850 Dec 17 11:54 logo-small.png
drwxrwxr-x 2 root     root      4096 Dec 17 12:52 meteor
drwxr-xr-x 2 pi       pi        4096 Dec 17 11:54 Model
-rw-r--r-- 1 pi       pi         108 Dec 17 11:54 passes.php
-rw-r--r-- 1 pi       pi        1509 Dec 17 11:54 style.css
drwxr-xr-x 2 pi       pi        4096 Dec 17 11:54 Views
pi@noaa1:~ $ 

pi@noaa1:~ $ ls -al /var/www/wx/meteor/
total 8
drwxrwxr-x 2 root root 4096 Dec 17 12:52 .
drwxr-xr-x 9 root root 4096 Dec 17 12:52 ..
pi@noaa1:~ $

reynico, 2020-12-17:
That's weird. Seems like some folders were created (and/or their ownership changed) to root.

```
chown -R pi:pi /var/www/wx/
```

Should be enough.

tomamdy, 2020-12-17:
I try it now

tomamdy, 2020-12-17:
The permission was root and now anyone. Shgould be better, cross my fingers.
Thanks

Tom.

tomamdy, 2020-12-17:
pi@noaa1:~ $ chmod -R 775 /var/www/wx/meteor/audio/
chmod: cannot access '/var/www/wx/meteor/audio/': No such file or directory
Only this seems strange for me.

reynico, 2020-12-17:
That's expected, as long as the directory doesn't exist (yet) but it'll created whenever a new pass decoding process is running.

tomamdy, 2020-12-17:
Thank You, that means that I have to be lazy.
Thanks, I have learned alot. For a men of nearly 73.

Tom

tomamdy, 2020-12-17:
Works very well. Thanks.

Tom.

closed 2020-12-19 by tomamdy