reynico/raspberry-noaa #80: [enhancement] Built-in monitoring dashboard

pineapplee85, 2020-12-18:
Hi there,

Just wondering what you thought about this idea?

Wondering if you could have a dashboard on the webpage (separate page) that shows stats such as CPU Usage and temp, WiFi strength, pass signal strength, satellite health eg. uptime or battery temp (if available from an external source?), cpu clock speed or any other data that might be useful in cool graphs and dials? Maybe something like this:
[https://jorge.maroto.me/content/images/2020/06/Screenshot-2020-06-08-at-00.59.38-1.png](url)

It would be cool to have it built into the script as a yes or no if you want it to install, however I understand this would require adding in lots of other steps and files into the script and may not be worth your time.

Thanks for the great project!
Ethan

reynico, 2020-12-19:
Hi Ethan!
Looks great but also resource-intensive. That's [Grafana](https://grafana.com/), it's an analytics/monitoring solution, really cool for sure but I want to keep this system as simple as possible. Anyways I can help you out installing this on your device so you can use it.

mihajlo2003petkovic, 2020-12-21:
Hi Nico! I would also like to install Grafana! Be sure to post an installation guide!

pineapplee85, 2020-12-22:
Hi Reynico,

I had a look at this but it did not seem to work, I also then had a go at node-red but to no joy. 
If you could help me that would be great.

Thanks!

reynico, 2020-12-24:
https://grafana.com/grafana/download?platform=arm

```
wget https://dl.grafana.com/oss/release/grafana-7.3.6.linux-armv6.tar.gz
tar -zxvf grafana-7.3.6.linux-armv6.tar.gz
```

then 
```
cd grafana-7.3.6
./bin/grafana-server
```

should be enough.

pineapplee85, 2020-12-24:
Hi there,

I have Grafana installed however, am not sure how to setup a dashboard. This is what I was confused with. 
Is there a tutorial? I could not find one anywhere.

Thanks!

reynico, 2021-01-03:
This is one of the best places to learn/ask about grafana I guess: https://www.reddit.com/r/grafana/

Since I'm not planning to add enhanced monitoring to raspberry-noaa I'll close this issue for now. Feel free to reopen if needed

closed 2021-01-03 by reynico