reynico/raspberry-noaa #81: Thinking about te software.

tomamdy, 2020-12-19:
Nico:
Thinking about te software.
Meteor works nice too.But where is the rectified image 
or is the image already rectfied?
How (whichcommands)could I change the RGB or color of te Meteor picture?
Where could I find the description of this commands?

But how many thumbnail are max. displayed in the webpannel?
What happened if the sd-card is full of pictures?

reynico, 2020-12-19:
Hi Tom! 

> Meteor works nice too.But where is the rectified image or is the image already rectfied?

The image is rectified during the reception, then the original image is deleted. Comment or remove this line so you keep both files: https://github.com/reynico/raspberry-noaa/blob/master/receive_meteor.sh#L82

> How (whichcommands)could I change the RGB or color of te Meteor picture?

For night passes, line 71: https://github.com/reynico/raspberry-noaa/blob/master/receive_meteor.sh#L71. For daylight passes, line 75: https://github.com/reynico/raspberry-noaa/blob/master/receive_meteor.sh#L75

> Where could I find the description of this commands?

 There is not much documentation yet, I'll try to write a bit this weekend.

> But how many thumbnail are max. displayed in the webpannel?

12 thumbnails per page, you can edit this value in `/var/www/wx/Config.php`, `img_per_page` setting.

> What happened if the sd-card is full of pictures?

Most likely, reception may stop working.

tomamdy, 2020-12-19:
**__Most likely, reception may stop working_.**_ But should it be possible to delete bij s.w. pictures and bring passe back to first ?
Recaption may stop workimg is not so nice solution.

Tom

reynico, 2020-12-20:
You mean an auto-prune process? Sure, it's easy to implement. I personally prefer keeping all my images and stop receiving new ones if the SD card is full instead of delete old ones to make more space. Using a 16GB SD card allows more than 8000 pictures!

tomamdy, 2020-12-20:
Oh YYes, I like a prune script. That learns me more.

Tom

reynico, 2020-12-20:
Prune script updated, https://github.com/reynico/raspberry-noaa/blob/master/prune.sh and instructions here: https://github.com/reynico/raspberry-noaa/blob/master/WORKING.md#pruning

```
cd /home/pi/raspberry-noaa/
git fetch -pt
git checkout origin/master -- prune.sh
```

tomamdy, 2020-12-21:
Thank's Nico. It seems to work well.

closed 2020-12-24 by reynico