reynico/raspberry-noaa #82: Twitter

tomamdy, 2020-12-20:
Making the devejoper acount is not easy, because the explaination (and my language knowledge) is unclear, if I try to make it on Twitter.

Tom

tomamdy, 2020-12-21:
I have made a developer account, but then nothing for me. I have not enough knowledge about twitter. May be advice is very welcome.

Tom

mihajlo2003petkovic, 2020-12-21:
After you've made the developer account, now you have to make an application. After you've done that, you make API tokes and enter it in the script and copy it. It is explained here: https://github.com/reynico/raspberry-noaa/blob/master/INSTALL.md#set-your-twitter-credentials

Explanation for creating Twitter application: https://docs.inboundnow.com/guide/create-twitter-application/

tomamdy, 2020-12-21:
May be you could help. This part, I don't understand, because it is public for erveryone.
  _A list of the government or public sector entities that will have access to Twitter content, or information derived from Twitter content, under this use case.
    The specific use cases of your product or service by government or public sector entities._

mihajlo2003petkovic, 2020-12-21:
Could you post a screenshot?

tomamdy, 2020-12-22:
At the moment very difficult to send a screeprint, becasue I had send the form.

closed 2020-12-23 by tomamdy